sccn/eeglab #490: std_precomp_error

evastone97, 2022-06-06:
Hello

I have another small study with 18 subjects and 3 conditions and I tried to precompute component measures and I got this memory issue problem.

I am sharing the folder with the study: https://drive.google.com/drive/folders/1SdzzuZnfnrxiUHyUaRCrVsjSCrUikskH?usp=sharing

Can you please let me know if the problem is only in my pc or is something internal?

[image]

Kind Regards,
Evangelia

arnodelorme, 2022-06-21:
You might want to disable parallel computation in the EEGLAB preference. Let me know

closed 2022-06-21 by arnodelorme

evastone97, 2022-06-26:
Ι disable the option but I still have the same issue, slightly better becsause it stopped in another participant but still I dont have the complete time frequency analysis for all of them.  
This is the study link: https://drive.google.com/drive/folders/1sJRlWqtLbWEUEvuvr5i3CJ-Mn9anxLE5?usp=sharing