sccn/eeglab #492: pop_comperp doesnt give output - empty figure without warnings

evastone97, 2022-06-08:
Hello,

I tried to compare two conditions from two different datasets by choosing the: Sum/Compare comp. ERPs  and I fill the following

[image]

But it gives me an empty figure, and this is what I have from the command window:
[image]

What do you think it might be a problem?

arnodelorme, 2022-06-21:
Yes, empty figure is not good. This function was removed from the menu though and is not being used anymore. Use the STUDY ERP plotting instead.

closed 2022-06-21 by arnodelorme

evastone97, 2022-06-21:
I find out why is because I didnt put anything in the component subset box (I should have put [] if I want all the components or channels) and then is fine.

arnodelorme, 2022-06-21:
Good to hear.

evastone97, 2022-06-21:
I have one theoritical question on the use of this function if you can answer me from here. I would like to find the electrodes with ethe maximum activity for the specific clusters. So I thought by using this function for the channels I can see where I see the higher amplitudes but my main question is how can I link the channels with the components in order to find the best ones?

evastone97, 2022-06-21:
Thank you! yes its useful, but I have to have selected or laoded a specific dataset. If I want to find the max activity from a cluster in my study desing what can I do?

arnodelorme, 2022-06-21:
You can do the same with the individual components of the clusters for each individual or taking the average cluster maps for all individuals.

arnodelorme, 2022-06-21:
I think "electrode with the highest amplitude from eash subject" is good enough. The other requires scripting.