sccn/eeglab #500: spherical interpolation

arnodelorme, 2022-06-15:
Makoto: Maruti pointed me to this issue--EEGLAB's electrode interpolation seems to follow Perrin et al. (1989), but there is an update on this topic in 2015 by Kang et al. (2015). I attached both papers here for your reference.

Question to Arno: Does eeg_interp() use the regularization parameter lambda? I don't think it is discussed in Perrin et al., but Kang et al. says it was introduced by Perrin et al. I'm a bit confused.

Anyway, it seems if we use computationally more expensive solutions (lower m, higher n) than those originally suggested by Perrin et al. (1989), it seems to produce better results in terms of the impact on the cross-electrode phase analyses.

arnodelorme, 2022-06-15:
Thanks Makoto,

Question to Arno: Does eeg_interp() use the regularization parameter lambda? I don't think it is discussed in Perrin et al., but Kang et al. says it was introduced by Perrin et al. I'm a bit confused.

I have looked and I do not think we use the parameter lambda. It is in Perrin et al in the comments at the end.

The current code is

 C = pinv([Gelec;ones(1,length(Gelec))]) * values;

Could be changed to

 C = pinv([Gelec+eye(size(Gelec))*lambda;ones(1,length(Gelec))]) * values;

Since lambda is a "is small positive value added to the diagonal in the matrix solutions of (1) to regularize"

We could change the default to m=3 and n=50, add lambda, and even make the number publicly accessible to users in the interface.

There is another function  sphericalSplineInterpolate in clean_rawdata that uses n=200; m=4 and lambda = 1e-5
Kay has also worked on a minor modification.

Attaching the original papers and a white paper of an observation I made more than 15 years ago. We actually do not use the standard spherical interpolation. We use a different method to compute the difference in angle, and our solution is apparently more robust to interpolation outer boundaries. I asked Jason to look at the math, but he was as puzzled as I am.

[Archive.zip](https://github.com/sccn/eeglab/files/8911862/Archive.zip)

MakotoMiyakoshi, 2022-06-15:
Thanks Arno. This is a perfect explanation. Maruti must be very impressed! It's also good to know that you had that white paper from 2007, I appreciate your deep thought and insight on this issue.

According to Kang et al. (2015), m=3, n=50, lambda=10^-8 is recommended. I think it is a great idea to update eeg_interp() based on Kang et al. (2015). clean_rawdata's n=200 should be certainly better but probably an overkill.

arnodelorme, 2022-06-15:
The original code Christian used is from Jason D.R. Farquhar ***@***.***) apparently.

MakotoMiyakoshi, 2022-06-15:
> We were not overly concerned about execution time in contrast to better
accuracy.

Yes, I totally agree.

Makoto

marutimishra, 2022-06-17:
Also, since you are able to dig into this and have expertise, I was wondering what you think about this article [here](https://link.springer.com/article/10.1007/s10548-021-00844-2), that seems to suggest a neurophysiological approach to interpolation rather than mathematical approaches like SSI.

MakotoMiyakoshi, 2022-06-17:
Yup, I was thinking about the same idea. The accuracy depends on the
forward model though. As Nunez and Srinivasan (2006) argues, we don't know
what is a good forward model because tissue conductivity is poorly known.
Thicker the skull, more resistance--intuitively it should work like that,
right? But you get a worse result there.

marutimishra, 2022-06-19:
Ah, I see. make sense.
I was thinking if one can use CNNs to interpolate? and I did a brief search and found this article that exactly does that https://www.sciencedirect.com/science/article/pii/S0165027021000613
what are your thoughts?

MakotoMiyakoshi, 2022-10-11:
You can use any algorithm for interpolation, and I do believe that each
approach has its own strength. But to compare these strengths across
methods not only requires a careful designing of simulation studies but
also would be challenged by variations of scenarios. A part of ground-truth
signal that is specific to the location of the electrode definitely
includes individual differences of brain anatomy including functional
mappings. Using a population-level knowledge to interpolate individual
anatomical/functional differences has a fundamental limitation from the
beginning. So I don't have any strong opinion as to which method should be
used here. I need to see more empirical and simulation test results to form
my own recommendation.

arnodelorme, 2022-10-13:
Is there any modification we should make to the function?

MakotoMiyakoshi, 2022-10-17:
It depends on whether EEGLAB developer wants to implement updated/'better'
channel interpolation to EEGLAB or not. It's not that I am requesting the
fix/change. The current algorithm does not do any bad.

Makoto

arnodelorme, 2024-02-10:
OK, I have added an option to use the Kang method and also use custom values.

closed 2024-02-10 by arnodelorme