sccn/eeglab #502: A compiled (standalone) version fails to load XDF files

marles77, 2022-06-17:
### Description

The plugin for importing XDF files in a standalone version does not work. 

<br>

#### Steps to Reproduce

1. Choose Import data from the File menu
2. Choose plugins/From .XDF or .XDFZ file
3. Watch a popup with a pop_loadxdf() error message

<br>

#### Expected behavior:

An open file dialog shows up to choose an XDF file.

<br>

#### Actual behavior:
A modal pops up with an error message informing that pop_loadxdf() is not recognized.

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [*-*] |
| EEGLAB version | [*2021.1*] |

arnodelorme, 2022-06-21:
OK, thank you. We will address this in the next compiled release within a month.

Favorable0350, 2022-06-28:
I have reproduced similiar error loading xdf files in standalone, octave gui and matlab R2013b. Older matlab version so i have extracted plugin instead of using manager.

In my case error message is:

EEGLAB error in function eeg_load_xdf() at line 35:

Undefined function "load_xdf" for input arguments of type "char"

arnodelorme, 2022-07-23:
Yes, it is a path issue. Will be addressed within 2 weeks