sccn/eeglab #503: Dot indexing is not supported for variables of this type (EEGLAB Loreta)

wafahabib5, 2022-06-23:
I wanted to create loreta using my dataset but then an error popped up regarding dot indexing not being supported by variables of this type

I loaded the dataset in EEGlab. It is a .matfile. I did ICA decompostion.
Then I wanted to do loreta in EEG after which this error message appears. My matlab version is R2021a. My EEGlab is 2021 version.

[image]

| OS version     | [Windows10] |
| Matlab version | [R2021a] |
| EEGLAB version | [2021] |

arnodelorme, 2022-07-13:
Apologies for the late response. I was on vacation. Would you mind sharing your dataset and instructions on how to reproduce the problem?

arnodelorme, 2022-07-23:
Closing this for now, but feel free to reopen the bug.

closed 2022-07-23 by arnodelorme

wafahabib5, 2022-07-30:
Hi,
I have solved the issue. I have another problem. Can you please guide me on
how to plot amplitide vs frequency graph for EEG signal? I have to make
comparison between EEG signal of controls vs drug group by analysing the
peak alpha frequency. Please assist me. I would really appreciate it. My
report is due this month. Is the power spectrum vs frequency graph the same
as amplitude vs frequency graph because I am able to do the power spectrum
using EEGlab. Please kindly help me.

Thank you.

Best Regards,
Wafa