sccn/eeglab #505: Epoch rejection error - eeg_checkset

Neetha-Das, 2022-06-24:
### Description

Hello! I use EEGLAB v2022.0. When I try to reject epochs automatically ( Tools->Automatic epoch rejection, or Tools-> Reject data epochs -> Reject extreme values) I get the following error: 

Error using [EEG LASTCOM] = eeg_checkset(EEG, 'epoch');if \~isempty(LASTCOM), if
LASTCOM(1) == -1, LASTCOM = ''; return; end; end; eegh(LASTCOM);try,[~ LASTCOM] =
pop_eegthresh(EEG, 1);catch, eeglab_error; LASTCOM= ''; clear EEGTMP ALLEEGTMP
STUDYTMP; end;EEG = eegh(LASTCOM, EEG);
Error: An array for multiple LHS assignment cannot contain expressions.
 
Error while evaluating Menu Callback.
<br>

#### Steps to Reproduce

1. Open EEGLAB
2. Load an epoched EEG .set file
3. Tools->Automatic epoch rejection

<br>

#### Expected behavior:
Options to reject epochs show up - fields to enter limits for thresholding the epochs, field to enter the channels etc. 

<br>

#### Actual behavior:
Crash with the above error

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | Windows 10 |
| Matlab version | 2018b|
| EEGLAB version | 2022.0 |

Thanks in advance for your help!

arnodelorme, 2022-07-13:
Can you reproduce the problem with version 2022.0? Would you mind sharing a dataset?

arnodelorme, 2022-07-23:
Closing this for now, but feel free to reopen the bug.

yzp12138, 2023-05-17:
hello,i meet same bug,did you find any sloution?

arnodelorme, 2023-05-17:
I think it has been fixed in 2023.0 but would you mind confirming?

yzp12138, 2023-05-18:
> 

no,my eeglab version is the latest one,other function can run successfuly ,only this one didn't work,the error description is below
`错误使用 [EEG LASTCOM] = eeg_checkset(EEG, 'epoch');if ~isempty(LASTCOM), if LASTCOM(1) == -1, LASTCOM = ''; return; end; end; eegh(LASTCOM);try,[~ LASTCOM] = pop_eegthresh(EEG, 1);catch, eeglab_error; LASTCOM= ''; clear EEGTMP ALLEEGTMP STUDYTMP; end;EEG = eegh(LASTCOM, EEG);
错误: 不支持将函数输出赋给此表达式。
 
计算 Menu Callback 时出错。`

yzp12138, 2023-05-18:
> 

Steps to Reproduce
1.Open EEGLAB
2.Load an epoched EEG .set file
Tools->reject data epochs->reject extreme values

dongwq66631, 2023-07-18:
Hello！ I meet same bug.Do you find a sloution?

yzp12138, 2023-07-18:
it will occure in some version of eeglab，you can use this function in ［ALL method］,it worked in my system---- Replied Message ***@***.***>Date07/18/2023 16:50 ***@***.***> ***@***.***>***@***.***>SubjectRe: [sccn/eeglab] Epoch rejection error - eeg_checkset (Issue #505)
Hello！ I meet same bug.Do you find a sloution?

[
{
"target": "https://github.com/sccn/eeglab/issues/505#issuecomment-1639804958",
"url": "https://github.com/sccn/eeglab/issues/505#issuecomment-1639804958",
"name": "View Issue"
},
"description": "View this Issue on GitHub",
"publisher": {
"name": "GitHub",
"url": "https://github.com"
}
}
]

dongwq66631, 2023-07-19:
Thanks for the approach！It's useful.

arnodelorme, 2023-07-19:
This has been fixed in 0c8d8622a

closed 2023-07-19 by arnodelorme