sccn/eeglab #506: file not properly imported from bdf files with EEGlab multiple versions

ashermanoxy, 2022-06-29:
### Description

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

ashermanoxy, 2022-06-29:
When I try to open BDF files into eeglab, I get an error as follows:
Warning SOPEN: number of event onset (TYP=7F00) and event offset (TYP=FF00) differ (0,1)
Warning SOPEN: number of event onset (TYP=7F01) and event offset (TYP=FF01) differ (0,118)
Warning SOPEN: number of event onset (TYP=7F02) and event offset (TYP=FF02) differ (0,225)
Warning SOPEN: number of event onset (TYP=7F03) and event offset (TYP=FF03) differ (0,260)
Warning SOPEN: number of event onset (TYP=7F04) and event offset (TYP=FF04) differ (0,607)
Warning SOPEN: number of event onset (TYP=7F78) and event offset (TYP=FF78) differ (0,59)
Warning SOPEN: number of event onset (TYP=7F82) and event offset (TYP=FF82) differ (0,300)
Warning SOPEN: number of event onset (TYP=7F84) and event offset (TYP=FF84) differ (0,300)
Warning SOPEN: number of event onset (TYP=7F8C) and event offset (TYP=FF8C) differ (0,60)
Warning SOPEN: number of event onset (TYP=7F8D) and event offset (TYP=FF8D) differ (0,60)
Warning SOPEN: number of event onset (TYP=7F8E) and event offset (TYP=FF8E) differ (0,60)
Warning SOPEN: number of event onset (TYP=7F96) and event offset (TYP=FF96) differ (0,60)
Warning SOPEN: number of event onset (TYP=7F97) and event offset (TYP=FF97) differ (0,60)
Warning SOPEN: number of event onset (TYP=7FA0) and event offset (TYP=FFA0) differ (0,60)
Warning SOPEN: number of event onset (TYP=7FAA) and event offset (TYP=FFAA) differ (0,60)
Warning SOPEN: number of event onset (TYP=7FAC) and event offset (TYP=FFAC) differ (0,60)
Warning SOPEN: number of event onset (TYP=7FF0) and event offset (TYP=FFF0) differ (0,1)
Warning SOPEN: number of event onset (TYP=7FFA) and event offset (TYP=FFFA) differ (0,599)
Warning SOPEN: number of event onset (TYP=7FFE) and event offset (TYP=FFFE) differ (0,599)

I have read the other chains around this issue but I can't seem to toggle into a different way to load the data to get the correct event numbers? The event numbers are numbers like 65280, 65282, 65284, 65534 and I'm not sure what they correspond to. Thanks for your help.

arnodelorme, 2022-07-01:
This is strange. Would you mind sharing a data file. @aschoegl could you have a look?

ashermanoxy, 2022-07-01:
MS_DFI_Summer2022_1.bdf.zip
<https://drive.google.com/file/d/1RfwlHkoO-cRNjldjp6xbAvv2Uq4qmJc0/view?usp=drive_web>
Thanks so much for the reply. Yes, here is a data file that will not load
properly. I had to send it through google drive because of file size limits
so please let me know if there is a problem opening the file.

-- 

*Aleksandra (Sasha) Sherman, PhD *(*she/her*) | Associate Professor,
Department Chair

*Department of Cognitive ***@***.***|  T 323-259-2786
Office Hours Appointment Link
<https://calendar.google.com/calendar/u/0/selfsched?sstoken=UURkQlVhVHphSTBYfGRlZmF1bHR8Y2RkZmY4YjNjMTdiMjQxZmI3MzAyYjUxNTIxNThkZmE>
Letter of Recommendation Request Form <https://forms.gle/KT17TFf5JYHYawKT9>

*OXY*
*Occidental College*

1600 Campus Road  |  Los Angeles, California 90041-3314

ashermanoxy, 2022-07-08:
Hi, I'm just wondering if you've had a chance to open the file and help resolve this issue? Th
anks so much.

closed 2022-07-12 by ashermanoxy

schloegl, 2022-12-03:
I checked this file, and decoding the status channel with bdf2biosig_events(..., mode=7) seems to result in some meaningful eventtable. Please check.