sccn/eeglab #507: Assign Bin Error [Invalid Identifier]

MrvKrt, 2022-07-01:
Hi, I have a problem with the stage of EEG Lab, Assign Bin. After creating event, I receive this error message below: 

fprintf error
`Invalid file identifier. Use fopen to generate a valid file identifier. (Error occured in function createeventlist() at line 239)`

arnodelorme, 2022-07-01:
Would you mind providing more detail. Which dataset, which event? We need to be able to reproduce the problem. Thank you.

MrvKrt, 2022-07-03:
I have a gtec hd5 file.  I follow the order analysing my data;
Import file, select data and apply channel locations. Apply ERP Lab, Filter for EEG data.
Channel Data (scrolls) cleaning part, after run ICA. Reject data using ICA, remove components.
Create Bin List. Assign Bins then I receive the error message. Here's my the error:

[image]

arnodelorme, 2022-07-13:
This is an ERPLAB problem apparently. I suggest copying and pasting your issue at https://github.com/lucklab/erplab/issues

closed 2022-07-13 by arnodelorme