sccn/eeglab #509: Function naming conflict and ICLabel

arnodelorme, 2022-07-23:
One weird issue we've been having on her end is that ICLabel produces NaN for all class probabilities when ICLabel is run in sequence with the various processing steps (filtering/ASR/AMICA). However, when MATLAB is restarted, ICLabel works normally. Only when run in sequence with the various preprocessing steps does it not work.

arnodelorme, 2022-07-23:
Fieldtrip bug submitted https://github.com/fieldtrip/fieldtrip/issues/2080

jjurgiel, 2022-07-24:
Full details:

Recreated this issue using Matlab 2020b, EEGLAB 2021.1, Fieldtrip-lite20210601, EEGLAB prompted me to install Fieldtrip-lite20220714 when running the pop_multifit function (below), which I hadn't seen happen before.

`    
[~,coordinateTransformParameters] = coregister(EEG.chanlocs, ...
        'D:\Documents\eeglab2021.1\plugins\dipfit\standard_BEM\elec\standard_1005.elc', 'warp', 'auto', 'manual', 'off');
    
% Dipfit settings
EEG = pop_dipfit_settings( EEG, ...
    'hdmfile', 'D:\Documents\eeglab2021.1\plugins\dipfit\standard_BEM\standard_vol.mat', ...
    'mrifile', 'D:\Documents\eeglab2021.1\plugins\dipfit\standard_BEM\standard_mri.mat',...
    'chanfile', 'D:\Documents\eeglab2021.1\plugins\dipfit\standard_BEM\elec\standard_1005.elc',...
    'coordformat', 'MNI', 'coord_transform', coordinateTransformParameters,'chansel', 1:EEG.nbchan);

EEG = pop_multifit(EEG, 1:size(EEG.icaact,1),'threshold', 100);

EEG = iclabel(EEG);`

The pop_multifit in particular added a number of /external/ Fieldtrip folders to the matlab path. Running ICLabel gives NaN values, unless you remove /external/signal from the path prior to running ICLabel. Note that this set of code has been run numerous times without issue nor any modifications since then - the only thing different is the updated Fieldtrip plugin.

arnodelorme, 2024-02-09:
This has been fixed by removing external folders in Fieldtrip by default.

closed 2024-02-09 by arnodelorme