sccn/eeglab #510: Error Reading Neuroscan .dat channel location file

amthapar, 2022-07-27:
### Description

[I am requesting assistance with importing a Neuroscan .dat channel location file. I have attached a photo of the NuAmps40.dat file (see attached) that worked with previous versions of EEGLAB without any issue but it does not work with the current version of EEGLAB (EEGLAB 2021). ]

<br>
#### Steps to Reproduce

1. Load .cnt file
2. Use EEGLAB Channel Locations to import the .dat file as per instructions on EEGLAB wiki

<br>
#### Expected behavior:
[*Correct Reading of channel locations*]

<br>
#### Actual behavior:
[Get the following Error Message

IMPORTANT: After importing/modifying data channels, you must close
the channel editing window for the changes to take effect in EEGLAB.
TIP: Call this function directy from the prompt, ">> pop_chanedit([]);"
     to convert between channel location file formats

readlocs(): 'dat' format assumed from file extension
Reading file (lines): 10 20 30 40 40
Index in position 1 exceeds array bounds. Index must not exceed 40.

Error in readneurolocs (line 91)
            locs = locs(tmpind,:);

Error in readlocs (line 413)
       eloc = readneurolocs( filename );

Error in pop_chanedit (line 722)
                        [chans] = readlocs(tmpargs{:});
 
Error while evaluating UIControl Callback.
]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*2021b*] |
| EEGLAB version | [*EEGLAB 2021*] |
[Photo of .dat file.docx](https://github.com/sccn/eeglab/files/9201176/Photo.of.dat.file.docx)

arnodelorme, 2022-09-06:
Thank you for reporting and apologies for the late response. We will work on the issue.

arnodelorme, 2022-09-06:
I have clarified the tutorial. It works but you had to extract data epochs first which was not clear.

closed 2022-09-06 by arnodelorme