sccn/eeglab #516: EEGLAB error in function std_readdata() at line 300

changrita1114, 2022-08-09:
### Description

I have tried to plot ERSP from a STUDY design using EEGLAB 2021.1 with MATLAB R2020b. There are two groups and one condition in the design, with unique subjects in each group. However, there was an error from a pop-up window indicated "std_readdata(line 300) error" with "Cannot handle conditions with different number of components for a given subject, This is not a bug (Error occurred in function std_readdata() at line 300)." I understand this error generally comes when there is a different IC decomposition within subjects (between conditions), but here we are using 100% unique subjects from two different diagnostic groups.
[The STUDY design picture](https://drive.google.com/file/d/1axScKnskrqXh7E-ZkU2m8W-hZp0N4kVD/view?usp=sharing)
[The error picture](https://drive.google.com/file/d/1Bv7B3KSL2qCkjXQh6Tf31L7vEUYykFP2/view?usp=sharing)

One observation (may or may not be related) is that ERSP plotting can work correctly if there is only one IC per subject in a cluster. Any idea or input would be appreciated.

The general steps I've taken are:
1. Load datasets into memory
2. Create study using loaded datasets
3. In STUDY gui, specify group of each subject
4. Precompute ERSP
5. Precluster/cluster/attempt to plot ERSP for a given cluster

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*macOS Monterey Version 12.5*] |
| Matlab version | [*R2020b*] |
| EEGLAB version | [*2021.1*] |

arnodelorme, 2022-09-06:
Apologies for the late reply. Yes, the error does not seem right. Unfortunately, I would need to see your STUDY to be able to track the problem. Any chance you can share it, even if it is really large?

stellafychang, 2022-09-15:
I uploaded our STUDY to a google drive and shared a link to your UCSD email address. Please let me know if you haven't received the line. Thanks.

arnodelorme, 2022-09-15:
OK, thanks. We will have a look.

Arno

AlejandroCallara, 2022-10-25:
Dear eeglabers,

I encountered a similar issue. 
I think I solved by replacing some lines in the code. I hope it helps!

Here you can check my std_readdata.m 

you can check the for cycle length at line 286 of my std_readdata.m:
for iDat2 = 1:length(dataTmp{iDat1}(:))
                  
the assignment of line 288:
dataTmp{iDat1}{iDat2} = double.empty(0,0,0); % sometimes empty but all dim not 0

the condition of if statement at line 299:
if any(any(compNumbers)
 
[std_readdata.zip](https://github.com/sccn/eeglab/files/9859078/std_readdata.zip)

stellafychang, 2022-11-23:
I just shared the link for the datasets. Please let me know if you didn't receive the link through your UCSD email. Thank you.

arnodelorme, 2023-03-24:
Thank you, I have added the fix. ccb275c9b

closed 2023-03-24 by arnodelorme