sccn/eeglab #517: pop_delset

arnodelorme, 2022-08-13:
When there is only one dataset, it refuses to clear it

dellabellagabriel, 2022-08-19:
If I understand correctly you're saying that `ALLEEG = pop_delset(ALLEEG, 1)` doesn't clear the fields when `ALLEEG` has one element? If that's the case I wasn't able to reproduce the bug. Maybe I misunderstood the problem.

arnodelorme, 2022-08-19:
Good point. Maybe it works from the command line, but not from the GUI.

dellabellagabriel, 2022-08-20:
The problem seems to be that the callback for the button "Delete dataset(s) from memory" eventually calls `eeglab redraw` which removes elements of `ALLEEG` that are empty:
```
% remove empty datasets in ALLEEG
while ~isempty(ALLEEG) && isempty(ALLEEG(end).data)
    ALLEEG(end) = [];
end
```
When there is a single element, `ALLEEG` will end up as a 1x0 struct which is eventually set equal to `EEG` in the `updatemenu` function:
```
if isempty(ALLEEG) && ~isempty(EEG) && all(arrayfun(@(eeg) ~isempty(eeg.data), EEG))
    ALLEEG = EEG;
end 
```
I see two ways of fixing this: either we don't remove cleared elements from `ALLEEG` or we don't set `ALLEEG` equal to `EEG` if it's empty. I'm not sure what the best approach would be in this case.

arnodelorme, 2022-08-21:
Yes, good point. This is a special case. I think it is better to output an error message instead of hacking the code. There is an alternative in the File menu. I have committed some changes in 279ca9e5d

Feel free to look at other bug reports and even issue pull requests. Your contribution is most welcomed.

```
	error( [ 'Cannot delete dataset when there is only one of them. Restart EEGLAB' 10 'or use the menu item in the File menu to clear all datasets' ]);
```

closed 2022-08-21 by arnodelorme