sccn/eeglab #528: Errors reported during ERSP plot

Naba1210, 2022-09-09:
### ![Errors reported during ERSP plot*]
(https://user-images.githubusercontent.com/74348971/189382765-49fe1a66-0780-4e09-accd-4730af65cace.jpg)
[image]

<br>
#### 
[image]
After precomputing, the data was further used for preclustering, where it reports of missing datasets. Replacing NaN values solved the purpose, however, while plotting th clusters multiple errors occured.

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2022-09-23:
Apologies for the late reply. Would you mind sharing your study (even if it is big).

arnodelorme, 2022-10-13:
Closing for now. Feel free to share your data and reopen.

Naba1210, 2022-10-13:
Hello
I have shared the data through another email id. Please confirm if
you received it or not.

Regards
naba sami

Naba1210, 2022-10-13:
---------- Forwarded message ---------
From: Naba Sami ***@***.***>
Date: Fri, Sep 23, 2022, 10:07 PM
Subject: EEG data ERSP analysis errors
To: ***@***.***>

Greetings Dr Arnaud,
This is Naba Sami, a PhD candidate from the Stress and Cognitive
Electroimaging Laboratory, currently working under Prof. Ratna Sharma. I
plan to use my EEG data of 60 epochs (4 sec each) acquired during a task
for ERSP Analysis by creating a study in EEG lab, for 5 subjects to compare
between two conditions. The steps were executed by referring to the EEGLAB
tutorials
https://eeglab.org/tutorials/10_Group_analysis/component_clustering_tools.html

There were a few errors initially, but later it was sorted. The study was
created for 5 subjects at the moment, but after precomputing for all the
subjects, the clustering could not be performed.
(Error: Do not have ICA pairs. Look for NaN values in study.cluster(1).sets
indicate missing datasets.)

The data structure was similar across all subjects and both conditions. I
fail to understand what needs to be done to rectify this issue. Could you
please take a look at the screenshot and help?
I am hereby attaching the screenshot for reference.
I shall be really grateful. Thank you in advance.

 Sub2_Blk1.fdt
<https://drive.google.com/file/d/1pIylHjFgyBpOWN4W3omaU055WaqMlpgq/view?usp=drive_web>

 Sub3_Blk1.set
<https://drive.google.com/file/d/1I1I_f6QK0_FnLdIFM0mqixXXg9hspAtA/view?usp=drive_web>

 Sub3_Blk1.fdt
<https://drive.google.com/file/d/1TNBhILTezRqiuTr7oCorGwBJcCUe1Ua5/view?usp=drive_web>

 Sub4_Blk3.fdt
<https://drive.google.com/file/d/1i_1dEPA80f8WZ_iLqc7YLJq_uSIEns6E/view?usp=drive_web>

 Sub4_Blk3.set
<https://drive.google.com/file/d/13T-Xpbv2mtUGOBCmbbG2QJoG9Rw7xfwI/view?usp=drive_web>

 Sub5_Blk2.set
<https://drive.google.com/file/d/1QtzxxhciZURd6cuvhyIFszl3Llj8vWsk/view?usp=drive_web>

 Sub1_Blk1.set
<https://drive.google.com/file/d/11xa8R_nxYQTCHmiUh_9rMCTrVVHqlO9w/view?usp=drive_web>

 Sub1_Blk3.set
<https://drive.google.com/file/d/1QEk68E7oTT582mvpaXMbOm9Rx30t2aT7/view?usp=drive_web>

 Sub5_Blk3.set
<https://drive.google.com/file/d/1z_qRj1PTc3cDNLJlwu2Sb4MBvvNLZMlu/view?usp=drive_web>

 Sub5_Blk1.fdt
<https://drive.google.com/file/d/1-VVeADUz2aSni_iKjmY9O4_yplqbCZIv/view?usp=drive_web>

 Sub5_Blk2.fdt
<https://drive.google.com/file/d/1R74XET91lEQVmXgjowd1dp-eGTOSrku1/view?usp=drive_web>

 Sub5_Blk3.fdt
<https://drive.google.com/file/d/1zZvLUn9fpshUYkLt6dQBvd0IdT8Cxxw3/view?usp=drive_web>

 Sub1_Blk3.fdt
<https://drive.google.com/file/d/1n_tCnMUNfb4wPqinqknhinnxpVN5wkBi/view?usp=drive_web>

 Sub1_Blk1.fdt
<https://drive.google.com/file/d/1q_e6JKoj_MCp10dZ-U1dCdxa01vjsSxd/view?usp=drive_web>

 Sub3_Blk3.set
<https://drive.google.com/file/d/1vyKqcrdP93uK78ezch81FASMi9w9qrbU/view?usp=drive_web>

 Sub3_Blk3.fdt
<https://drive.google.com/file/d/1l9piXyjkF1ZEhocryNRulHrfk2dK4-Id/view?usp=drive_web>

 Sub3_Blk2.set
<https://drive.google.com/file/d/1atdcWHZBnDpHhsXVJKP_16W6WBhJgSMQ/view?usp=drive_web>

 Sub4_Blk2.set
<https://drive.google.com/file/d/1goOLLmPI-c8YPoHmk8GZQ-qHjg90-KYl/view?usp=drive_web>

 Sub4_Blk2.fdt
<https://drive.google.com/file/d/18cTWNYHsxl3JsKOK77QN4XRX7IgPUccq/view?usp=drive_web>

 Sub2_Blk2.set
<https://drive.google.com/file/d/1gEU3IfjBjQ5OqfZMlxq9ZWCvnj7KvqaY/view?usp=drive_web>

 Sub2_Blk2.fdt
<https://drive.google.com/file/d/1wjQ-lHJRCgFkzhYi_qkw4wEmzxEmhRxf/view?usp=drive_web>

 Sub2_Blk3.fdt
<https://drive.google.com/file/d/1EiwWj_Qos0NvPjyeU8Dw6MasYrdkhFk8/view?usp=drive_web>

 Sub2_Blk3.set
<https://drive.google.com/file/d/1ZsThXqEkQSlFf82tP36QMkro21h-Osza/view?usp=drive_web>

 Sub3_Blk2.fdt
<https://drive.google.com/file/d/11e22yuBO4G6wU8cOp5_hQJsyNyc_wRNe/view?usp=drive_web>

 Sub2_Blk1.set
<https://drive.google.com/file/d/12LIlC80zSJ5Xc1BI_TdcENJ9_IIvZ4_3/view?usp=drive_web>

 Sub4_Blk1.set
<https://drive.google.com/file/d/1_VilDdCdeR3tuu2pu_IQPQ3LJMxmjh9L/view?usp=drive_web>

 Sub4_Blk1.fdt
<https://drive.google.com/file/d/1aXPXo5qWxJTE6HS8H6vLhliKPWW_GEBM/view?usp=drive_web>

 Sub5_Blk1.set
<https://drive.google.com/file/d/1EoO91IUJX5FQqnoohBKXYHdD9Omyz-Gw/view?usp=drive_web>

 Sub1_Blk2.fdt
<https://drive.google.com/file/d/1FsklC39mW1HRlS8hjcyvXmtypJECIZuH/view?usp=drive_web>

 Sub1_Blk2.set
<https://drive.google.com/file/d/1wzVKcZsWcWBXgB_7BWjC64bNXuai9XQ4/view?usp=drive_web>
-- 
*Regards*

*Naba Sami*

*Ph.D. Scholar *
*Stress and Cognition Electroimaging Lab (SCEL)*
*Department of Physiology*
*All India Institute of Medical Science (AIIMS)*
*Ansari Nagar East*
*New Delhi, India*

arnodelorme, 2022-10-13:
Thank you. I will need the study file as well. Can you share the folder instead. This way I can download it as zip.

Naba1210, 2022-10-13:
Thank you for your reply.
I will take sometime to share it in zip.

Regards
Naba Sami

arnodelorme, 2022-12-01:
Sorry I was traveling. You can post the data (7Gb) at https://drive.google.com/drive/folders/1d4h4W3JXcwEWQsUh3a2cFPlourieXZeF?usp=share_link

arnodelorme, 2023-03-24:
Closing this for now but feel free to share the data at the provided link, if you want us to have a look.

closed 2023-03-24 by arnodelorme