sccn/eeglab #529: Unrecognized function or variable 'NChannelsRec'

celialitovsky, 2022-09-09:
I am running Matlab 2021b, EEGLAB version 2022.1, with the Neuroeelctrics toolbox version 1.1 on Mac OS Big Sur 11.6.8 (Intel). When I try to upload my .nedf file, I get the error:
EEGLAB error in function pop_nedf() at line 203: Unrecognized function or variable 'NChannelsRec'. In my Matlab command window, I also get the below warning:
Warning: Name is nonexistent or not a directory: [path]/eelab2022.1/xml4mat.

Any help would be appreciated. Thanks!

celialitovsky, 2022-09-09:
I figured out the issue and wanted to share in case it helps someone else! The Neuroelectrics website has the original version of the toolbox (1.1) but there is an updated version within EEGLAB. Go to File > Manage EEGLAB extensions > update to newest version of Neuroelectrics toolbox. Works now!

closed 2022-09-09 by celialitovsky

arnodelorme, 2022-09-09:
Good to know. Thanks for the update.