sccn/eeglab #533: several psd-related functions fail with low sampling rate EEG data

jiversen, 2022-09-13:
### Description

A workshop participant had an extremely downsampled dataset with EEG.srate = 64. 

inspect/label components by map and ICLabel both failed.

inspect/label components by map is hardwired to calculate PSD up to 50Hz
ICLabel is hardwired to calculate PSD up to 100Hz

Thus, data with EEG.srate < 100 will fail for inspect/label components by map and data with EEG.srate < 200 will fail with ICLabel. 

### Solution

The easiest would be to add a check of EEG.srate and issue a warning that the data is not suitable.

arnodelorme, 2022-09-23:
Good point. I will add a warning message and maybe remove the constraint.

arnodelorme, 2023-03-24:
From https://github.com/sccn/ICLabel/issues/31

It appears that ICLabel creates PSD features with max frequency 100 Hz, and thus needs data with sampling rate >= 200 Hz.

However, ICLabel is written to assume that the only way it could fail is if the model mex has not been compiled properly, and in this case recompiles the mex file, which is the wrong solution. Then it tries again and fails again, incorrectly warning that complication was unsuccessful, while it was in fact successful.

In this case the model input is the problem, not the model. This is very confusing for the user.

Solution: add a check for EEG.state < 200, and error out informatively if so.

arnodelorme, 2024-02-09:
OK, an error is being shown now.

closed 2024-02-09 by arnodelorme