sccn/eeglab #537: clean_rawdata issue

marutimishra, 2022-10-03:
Hi,
The Cleanrawdata gui throws this error if I use it on specific channels. If I donot do channel selection, it works fine
here is the file with the screenshot of selections from gui
https://drive.google.com/drive/folders/1KfrXo9M_RTQEYBQ4je_iBQy7LPaPOIRz?usp=sharing

arnodelorme, 2022-10-04:
I have fixed the issue which was due to the fact that you already had discontinuity in your data. You can pull changes from Github (clean_rawdata) folder. Alternatively, you can call the clean_rawdata twice. First remove channels and then remove data. This should work as well.

closed 2022-10-04 by arnodelorme

marutimishra, 2022-10-05:
Hi,
Now when i do it, it works but just that all the event information is lost. Did you notice that?

arnodelorme, 2022-10-05:
Let me check

arnodelorme, 2022-10-05:
I cannot reproduce your new problems. The events are there for me.

Arno

marutimishra, 2022-10-05:
Right, so if i chose a previous file type which is now continuous it throws that error. So wont be doing that. 
But if I just select one option from the gui (for detecting bad channels, with parameters to ignore few channels and then giving thresholds for those options in bad channel marking) and I deselct everything else--except the plot to compare tool I get the below error... any idea?

Warning: vis_artifacts failed. Skipping visualization. Could be because of duplicate channel label. 
> In pop_clean_rawdata (line 218)