sccn/eeglab #545: Error reported for ERSP plot in Study

Naba1210, 2022-10-27:
ERROR: Number of elements must not change EEGlab study

Dear EEGLAB,

I am writting here to report a bug related to the std_stdcond function (also in the GUI).

#### Steps to Reproduce
 
Load the [Objective1.study]
Study location: https://drive.google.com/drive/u/2/folders/0ALSJMw_UHlf9Uk9PVA
[image]

[First Step]
STUDY-->Run ICA-->Head Model and fitting-->Component Dipole autofit-->DIPFIT<15%

[Second Step]
STUDY --> Precompute channel measures -->
[Third Step]
STUDY -->Pre clustering
[image]

[Fourth Step]-->cluster complonents

[Fifth Step]--> Plot ERSP [ERROR]

Number of elements must not change. Use [] as one of the size inputs to automatically calculate the appropriate size for that dimension.

[image]

Hypotheses
I suspect that the data structure has similar events but there are some files which are showing uevents "[]"

The original files are all having same data structure, but turns out after precomputing this was the only difference i could find

<html>
<body>

OS version | Windows 11 22000.1098
-- | --
Matlab version | Matlab R2021b
EEGLAB version | EEGLAB v13.6.5b

</body>
</html>

arnodelorme, 2022-12-01:
Apologies for the late reply. I was traveling. I have requested access to the data.

arnodelorme, 2023-03-24:
The shared folder is empty so I cannot test. Closing this report for now.

closed 2023-03-24 by arnodelorme