sccn/eeglab #559: eeglab error in the function getfilename() at line 504---STUDY >plot erps

MJ202201, 2022-11-29:
### Description

When I try to use STYDY plot ERPs 

I got an error saying **_"eeglab error in the function getfilename() at line 504:
Array indices must be positive integers or logical values"_**

I have tried to debug it by myself but apparently, I cannot find the function named getfilename()... So now I do not know where to start

Can you help me to fix this problem?

The step I went through is

1. File > Load existing STUDY 
2. Edit > channel location
3. Study > precomputed data
4. Study > plot channel measures > plot erps

Thank you in advance
\

arnodelorme, 2022-12-21:
Apologies for the late response. WOuld you mind sharing your entire study?

mdussaultgomez, 2023-02-03:
Hi, was there ever a solution to this issue? It's happening to me as well whenever I try add new subjects to an already created study.

leima1, 2023-04-03:
I am also unable to add new subjects/runs to an existing study. 

EEGLAB returns error "Unrecognized function or variable 'realindex. ' 
In function std_editset() at line 292.

It appears that realindex is created at line 169, if STUDY does not have dataset info. However, if the study has dataset info, then std_editset crashes at line 292.  

Swapping 'realindex' with 'currentind,'  at line 292 seems to help, but I want to double-check if there are any consequences in doing that. @arnodelorme 

I am using eeglab v2023.0

Thanks!

arnodelorme, 2023-05-12:
Good. There is no onsequences in doing that. I have fixed the code.

YANKI21, 2023-07-21:
Dear arnodelorme,

I am also taking the same error, line 504 getfilename. 

Could you please help?

arnodelorme, 2023-07-21:
Would you mind trying with the latest EEGLAB version?

arnodelorme, 2024-02-09:
OK, the cache has been cleared as well when adding a new dataset.

closed 2024-02-09 by arnodelorme