sccn/eeglab #562: LIMO error

jsasisekaran, 2022-12-01:
Hello,

I am receiving this error message when I try to run LIMO after computing preclustering for component and channel measures.

See attached error msg.

I can send the study file if it might help with fixing the issue.

<br>
#### Expected behavior:
I expected that I would be able run LIMO on a repeated measures design with 4 levels of Factor 1.

<br>
#### Actual behavior:
See attached error msg.

says line 326; Index must not exceed array elements; index must not exceed 3.
<br>
All recent version.

|   |   |
|----| ---- | 
| OS version     | [*11
| Matlab version | [2022a] |
| EEGLAB version | [2022|

[image]

arnodelorme, 2023-03-24:
Can you share the file with @CPernet so he can have a look?

CPernet, 2023-03-24:
I sent a request to access .... On top of my head an error in std_limo indicates you are trying to set some repeated measurements via studyjich not how it works. One file per subject only to run a GLM and then do a repeated measure ANOVA on the models parameters.

jsasisekaran, 2023-08-01:
Hi Arnold, Cyril,

I am getting back to this issue again. I did a trial run with 8 participants. I got through till Limo Level 1 analysis. I am receiving an error at Level 2 analysis. The design is reduced from a repeated measures (1 factor 4 comparisons) to comparing for a syllable effect (1st vs. second syllable) in ERSP. I can share the study file, I'd appreciate your help with this.

Thanks,
Jay

CPernet, 2023-08-01:
1st let's make your 1st level correct -- you have 4 conditions right? so does your design matrix have 5 columns? (4 conditions plus the constant term)

jsasisekaran, 2023-08-01:
>>1st let's make your 1st level correct -- you have 4 conditions right? so does your design matrix have 5 columns? (4 conditions plus the constant term)

The visual matrix has 5 columns. LIMO (4.0) provides 4 conditions and 2 constants, maybe because of the initial design (4 conditions) and the newer with 2 conditions (1 vs. 2nd syllable)? I have shared the files on google drive (sasis001@umn.edu). I tried both repeated measures with 4 conditions and paired t with 2 conditions. Both give errors at Level 2.

Thanks for your help with this!
Jay

jsasisekaran, 2023-08-20:
Hi Cyril (This might be a question for @arnodelorme),

My intention was a repeated measures comparing all 4 contrasts. LIMO is not working for, re: 2nd level tests. Hence, I am trying to retrieve the data from matlab and bringing the data into a statistical program for further analysis. After precomputing channel measures with the default parameters--Using 3 cycles at lowest frequency . Generating 200 time points from -442.4 to 1440.4 ms.

Then I use this to obtain the output files (I gathered this info. from the listserv).
STUDY  = pop_erspparams(STUDY, 'freqrange', [alpha/theta frequency] , 'timerange', [0 1400]);
[STUDY erspdata ersptimes erspfreqs] = std_erspplot(STUDY, ALLEEG, 'channels', {'F7', 'F8', 'AF7', 'AF8', 'TP7', 'TP8', 'PO7', 'PO8'}, 'plotsubjects', 'on');

The ERSP output files have the required information except I am considering if this data is baseline corrected or not? Could you say if I should apply baseline correction on the data. If yes, is there a recommended protocol?

Thanks,
Jay

CPernet, 2023-08-20:
It's a little frustrating to state 'IMO is not working for, re: 2nd level tests' and not explaining what is not working -- from the data you shared, I could do t-tests and ANOVA on the 8 subjects.

arnodelorme, 2024-02-09:
@CPernet can we close?

CPernet, 2024-02-09:
Yes

closed 2024-02-09 by arnodelorme