sccn/eeglab #566: Additions change order of fields on EEG.chanlocs

nucleuscub, 2022-12-02:
Hi @arnodelorme , this is key for next week's event. 

The lines below are changing the default order of the fields on EEG.changelocs. Therefore, many of the functionalities that rely on the order of the fields are failing now. E.g., pop_chanedit with 'append'.

I saw you reverted this change in the development branch. Any chance we can make a release with this change and others we can take benefit from (e.g., the one in supergui I recently submitted)?
Thks, 
R

https://github.com/sccn/eeglab/blob/cc284f106bf9be65db3a32cccf82f748d80bfd22/functions/popfunc/pop_fileio.m#L236-L241

arnodelorme, 2022-12-02:
Yes, of course. I think we can move this piece of code at the end of the “if” statement, it will not change the order of the field. 
If you have an example that does not work, I can fix it easily.

nucleuscub, 2022-12-03:
Thanks, Arno. Yes, will see this happening e.g., when importing the fif file we will use for the practicalMEEG (sub-01_ses-meg_task-facerecognition_run-01_proc-sss_meg.fif). This is located in PracticalMEEG_2022/Data/sub-01. This is the Google Drive folder that Johanna shared with you.
When loading this file you will see the order of the fields in EEG. chanlocs have the field 'type' in an unusual position. 
Thnks,
R

arnodelorme, 2022-12-04:
I loaded the file and type looks good (at about the end of the channel location structure).

arnodelorme, 2022-12-04:
[image]

arnodelorme, 2022-12-04:
Let me know what I got wrong.

nucleuscub, 2022-12-04:
Thanks for taking a look into this, Arno. Let's better take a look together. After pulling the changes, I'm getting the right order now, but no values for channel locations at all. :/

arnodelorme, 2022-12-04:
Sounds good. Let’s do that on Tuesday

arnodelorme, 2023-03-24:
@nucleuscub can you provide an example to illustrate the problem?

arnodelorme, 2024-02-09:
Closing that for now. @nucleuscub please reopen if you have a chance. It is important to get that right.

closed 2024-02-09 by arnodelorme