sccn/eeglab #568: LIMO Yr Not Computing for Components

edwardwl, 2022-12-07:
Hello, 

I am having some issues with LIMO not estimating model parameters (components). I read another Github thread that mentioned that it could be not computing Yr (https://github.com/LIMO-EEG-Toolbox/limo_tools/issues/71) possibly due to data of different length/epochs. I tried this, and it also did not work. It does work for estimation of channel model parameters and regression analysis of channel data, just not for components. Also, similar to the other Github post, I can also see the ICs and clusters in the clustedit window.

Data:
Epoched resting state data that has been preprocessed.
Initial data was ~10 mins. 
Current data is 10 Epochs, 5 s each. 
Avg reference
ASR deletion of bad segments (K=40)
Downsampled to 250
DPFIT and IC (extended infomax) were used.
Link to sample data: https://drive.google.com/drive/folders/1dFIdOn3_teWO-epCVIsXyVBc_0gJr5hs?usp=share_link

Precluster/Cluster:
Precomputed Components W/ Spectra & Scalp
Preclustered W Spectra
Clustered W/Affinity

Model:
Design: 1st lvl variables: Constant
Input data to use for the GLM: Spectrum
Optimization method: WLS or OLS (have used both with same results)
Options:' freqlim', [1 25]

The error reads:
Something went bad ... the job has FAILED !
The last error message occured was :
Reference to non-existent field 'cellArray'.
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\functions\popfunc\pop_loadset.m at line 284
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\functions\popfunc\pop_loadset.m at line 152
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\limo_batch_design_matrix.m at line 24
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_job.m at line 214
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_job.m at line 128
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_script.m at line 437
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_script.m at line 323
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_pipeline_process.m at line 591
File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_pipeline.m at line 417
File C:\Program Files\MATLAB\R2020b\toolbox\matlab\lang\parallel_function.m at line 832
File C:\Program Files\MATLAB\R2020b\toolbox\parallel\remoteParallelFunction.m at line 67

Something went bad ... the job has FAILED !
The last error message occured was :
Error using load
Unable to read file 'Yr'. No such file or directory.
06-Dec-2022 17:07:50 glm    submitted (1 run / 1 fail / 1 done / 0 left)

Versions:
LIMO version v3.3
| OS version     | Windows 10 Enterprise. Version: 22H2. OSbuild:19045.2251 |
| Matlab version | R2020b |
| EEGLAB version | "latest"|
You are using the latest version of EEGLAB.

arnodelorme, 2022-12-07:
@CPernet

CPernet, 2022-12-07:
so assuming the error message from matlab is correct, I fails @https://github.com/LIMO-EEG-Toolbox/limo_tools/blob/master/limo_batch_design_matrix.m#L24 i.e. `EEGLIMO=pop_loadset([LIMO.data.data_dir filesep LIMO.data.data]);`

if you go to a given subject and `load('LIMO.mat')`, is the path/file listed by `[LIMO.data.data_dir filesep LIMO.data.data]` correct? and if correct can you confirm that  `EEGLIMO=pop_loadset([LIMO.data.data_dir filesep LIMO.data.data]);` fails despite the path to file being correct?

thx

edwardwl, 2022-12-08:
Thank you for the response! Yes, I can confirm that [LIMO.data.data_dir filesep LIMO.data.data] provides the correct file path and  EEGLIMO=pop_loadset([LIMO.data.data_dir filesep LIMO.data.data]) fails to load. LIMO does created the *_ses-1_design1.set files, but I cannot open them.  I uploaded the files to the google drive.

This is what it shows:

_

> >> EEGLIMO=pop_loadset([LIMO.data.data_dir filesep LIMO.data.data])
> pop_loadset(): loading file D:\HEALEEG\LIMOTest\sub-HEALEEG-BIO003-00-REC_LIMOTest_ses-1_design1.set ...
> pop_loadset(): Incompatible with new format, trying old format and converting...
> Reference to non-existent field 'cellArray'.
> 
> Error in pop_loadset>checkoldformat (line 284)
> 		eegset = EEG.cellArray;
> 
> Error in pop_loadset (line 152)
>             EEG = checkoldformat(TMPVAR.EEG);

_

I also saw this in the prior error that may mention a possible lack of saving of preclustering.
  
```

>   ******************************
>   The job starts now !
>   ********************
>   single matrix, nothing to concatenate
>   
>   ********************
>   Something went bad ... the job has FAILED !
>   The last error message occured was :
>   Error using limo_batch_design_matrix (line 159)
>   to run component clustering, you need the EEGLAB study loaded in the workspace with the clustering computed and saved
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\limo_batch_design_matrix.m at line 159
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_job.m at line 214
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_job.m at line 128
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_script.m at line 437
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_script.m at line 323
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_pipeline_process.m at line 591
>   File C:\Users\edwar\Documents\MATLAB\eeglab2022.1\plugins\LIMO3.3\external\psom\psom_run_pipeline.m at line 417
>   File C:\Program Files\MATLAB\R2020b\toolbox\matlab\lang\parallel_function.m at line 832
>   File C:\Program Files\MATLAB\R2020b\toolbox\parallel\remoteParallelFunction.m at line 67

```

When I checked STUDY.preclust is empty but, STUDY.cluster.preclust is full for all CIs except parent

CPernet, 2022-12-08:
@arnodelorme can you check what's wrong in the .set such as pop_load fails? those intermediate files are generated in std_limo.m ; I'm not sure why it fails, my guess is no condition+ICA is a case we have not tested I think

edwardwl, 2022-12-21:
Hello, I wanted to check in if there is anyway that I can help troubleshoot.

CPernet, 2022-12-21:
there might be later on an issue with STUDY.preclust vs. STUDY.cluster.preclust but at the moment the error is `EEGLIMO=pop_loadset([LIMO.data.data_dir filesep LIMO.data.data]);` @arnodelorme have you had a chance to look at this? I believe the intermediate ICA files are wrong

@edwardwl in between, you could also try to not use STUDY and simply do import in LIMO EEG and see if that works ; it should import the ICA files but you need to provide txt files for the different conditions (see https://github.com/LIMO-EEG-Toolbox/limo_tools/wiki/Defining-conditions) ; if this works (you could do just 2 subjects) we know for sure the prblm is stemming from intermediate STUDY files (the issue with this method of imåporting is that you loose the clustering info)

arnodelorme, 2022-12-21:
There is no data in the file. These should be named '.mat' not '.set' to avoid confusion.

edwardwl, 2022-12-22:
Hello, I think I have a big lead. The problem only arises after clustering. If I don't cluster, I don't get any errors when Estimating Parameters (components) through study.  Possibly clustering is not designed to function with only resting state data? Thanks for your help!

> ******************************
> Log of the (matlab) job : design
> Started on 22-Dec-2022 12:21:08
> User: edwar
> host : unknown
> system : windows
> ******************************
> The job starts now !
> ********************
> single matrix, nothing to concatenate
> computing design matrix
> making up the design matrix and data files ...
> 
> ****************
> Checking outputs
> ****************
> The output file or directory C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\derivatives\sub-3\TestLIMOFinal_design1_GLM_Components_Frequency_WLS\Yr.mat was successfully generated!
> 
> *********************************************************
> 22-Dec-2022 12:21:08 : The job was successfully completed
> Total time used to process the job : 0.20 sec.
> *********************************************************
> 
>     The feedback from the execution of job design was : 
> :22-Dec-2022 12:21:12 design completed (0 run / 0 fail / 2 done / 1 left)
> 22-Dec-2022 12:21:12 glm    submitted (1 run / 0 fail / 2 done / 0 left)
> 
>     The execution mode is session
>     The following command is going to be executed :
> psom_run_job('C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\derivatives\LIMO_TestLIMOFinal\limo_batch_report\TestLIMOFinal_design1_GLM_Components_Frequency_WLS\subject1\\glm.mat')
> 
> 
> ******************************
> Log of the (matlab) job : glm
> Started on 22-Dec-2022 12:21:12
> User: edwar
> host : unknown
> system : windows
> ******************************
> The job starts now !
> ********************
> analyzing component 1/65 
> analyzing component 2/65 
> analyzing component 3/65 
> analyzing component 4/65 
> analyzing component 5/65 
> analyzing component 6/65 
> analyzing component 7/65 
> analyzing component 8/65 
> analyzing component 9/65 
> analyzing component 10/65 
> analyzing component 11/65 
> analyzing component 12/65 
> analyzing component 13/65 
> analyzing component 14/65 
> analyzing component 15/65 
> analyzing component 16/65 
> analyzing component 17/65 
> analyzing component 18/65 
> analyzing component 19/65 
> analyzing component 20/65 
> analyzing component 21/65 
> analyzing component 22/65 
> analyzing component 23/65 
> analyzing component 24/65 
> analyzing component 25/65 
> analyzing component 26/65 
> analyzing component 27/65 
[... 145 lines cut ...]
> The output file or directory C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\derivatives\sub-3\TestLIMOFinal_design1_GLM_Components_Frequency_WLS\Betas.mat was successfully generated!
> 
> *********************************************************
> 22-Dec-2022 12:21:13 : The job was successfully completed
> Total time used to process the job : 1.26 sec.
> *********************************************************
> 
>     The feedback from the execution of job glm was : 
> 722-Dec-2022 12:21:17 glm    completed (0 run / 0 fail / 3 done / 0 left)
> 
> *********************************************
> The processing of the pipeline is terminated.
> See report below for job completion status.
> 22-Dec-2022 12:21:17
> *********************************************
> All jobs have been successfully completed.
> LIMO batch processing finished succesfully
> LIMO batch works thanks to PSOM by Bellec et al. (2012)
> The Pipeline System for Octave and Matlab. Front. Neuroinform. 6:7
> Re-saving study file
> All subjects have been successfully processed.
> Done.
> >> 

```

CPernet, 2022-12-23:
so @arnodelorme also use LIMO without experimental conditions for frequency analysis, cutting the EEG in N epochs -- can you tell us what you do here? then we can try to figure out what's up with the clustering

edwardwl, 2022-12-30:
Of course! This is what I use to extract epoch from the continuous data. 

EEG = eeg_regepochs(EEG, 'recurrence',5, 'eventtype','5','extractepochs','off');
EEG = pop_epoch(EEG, {'5'}, [-2.5, .2.5]);

I also tried to conduct my preprocessing only using the GUI and I had the same results. Was able to get lvl processing for channels. For components, it only functioned when I did not cluster. (GUI preprocessed .Set and .STUDY files uploaded) I can also attach my preprocessing script if it is helpful.

I also tried using the LIMO plug in Batch Upload and It would not upload unless I had its parent STUDY uploaded. After uploading its parent STUDY file, I was able to upload the files. However, I then received this failed job notice from LIMO. 

> Please use our boilerplate Citation and Reporting:
> https://github.com/LIMO-EEG-Toolbox/limo_tools/wiki/Reporting-methods-and-results
> References are in the citations.nbid file
>  
> C:\Users\edwar\Desktop\ReseachProjects\EEG\LIMO\Anxiety_Depression_Long\TwoTest.txt found 
> --------------------------------
> processing model 1/1 
> --------------------------------
> 
> opt_init = 
> 
>   struct with fields:
> 
>          path_logs: 'C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\'
>        path_search: (NOTE deleted this part of the message for brevity)
> 
> ***************************************************************************************************************************************************************************************************************
> The pipeline description is now being prepared for execution.
> The following folder will be used to store logs and status :
> C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\
> ***************************************************************************************************************************************************************************************************************
> Examining the dependencies of the pipeline ...
>     Removing empty jobs ...
>     Checking that all jobs are associated with a command ...
>     Generating dependencies ...
>        Reorganizing inputs/outputs ... 0.00 sec
>        Analyzing job inputs/outputs, percentage completed :  30 65 100- 0.00 sec
>     Checking if some outputs were not generated twice ...
>     Checking if the graph of dependencies is acyclic ...
>     Cleaning up job status ...
> 
> Setting up the to-do list. The following jobs will be executed ...
>     import (new)
>     design (new)
>     glm    (new)
> Checking if all the files necessary to complete the pipeline can be found ...
> 
> Saving the pipeline description in the logs folder ...
>     Creating the logs folder ...
>     Creating output folders ...
>     Removing old outputs ...
>     Creating a 'lock' file C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE.lock ...
>     Saving the individual 'jobs' file C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE_jobs.mat ...
>     Saving the pipeline dependencies in C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE.mat...
>     Saving the 'status' file C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE_status.mat ...
>     Saving the 'logs' file C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE_logs.mat ...
>     Saving the 'profile' file C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\\PIPE_profile.mat ...
>     Cleaning up old tags and logs from the logs folders ...
> 
> The pipeline has been successfully initialized !
> 
> opt_proc = 
> 
>   struct with fields:
> 
>                      mode: 'session'
>     mode_pipeline_manager: 'session'
>                max_queued: 1
>              qsub_options: ''
>             shell_options: ''
>            command_matlab: 'C:\Program Files\MATLAB\R2020b\bin\matlab'
[... 55 lines cut ...]
> 
>     The feedback from the execution of job import was : 
> :29-Dec-2022 16:12:20 import failed    (0 run / 1 fail / 0 done / 2 left)
> 
> *********************************************
> The processing of the pipeline is terminated.
> See report below for job completion status.
> 29-Dec-2022 16:12:20
> *********************************************
> The execution of the following job has failed :
> 
>     import ; 
> 
[... cut ...]
>     psom_pipeline_visu('C:\Users\edwar\Desktop\ReseachProjects\EEG\EEGLAB Bug-20221222T001208Z-001\EEGLAB Bug\Manual Test\derivatives\LIMO_LIMOManualTest\limo_batch_report\LIMOManualTest_Design2_GLM_Channels_Frequency_WLS\subject1\','log',JOB_NAME)
> 
> The following jobs have not been processed due to a dependence on a failed job or the interruption of the pipeline manager :
> 
>     design ; glm ; 
> 
> All jobs have been processed, but some jobs have failed.
> You may want to restart the pipeline latter if you managed to fix the problems.
> LIMO batch works thanks to PSOM by Bellec et al. (2012)
> The Pipeline System for Octave and Matlab. Front. Neuroinform. 6:7

CPernet, 2023-01-03:
ok, still that weird limo_batch_import_data.m at line 25 when ICA + cluster , that is crazy ... I'll see if I can reproduce this, thx for your patience 

in between .. since you have ICA working (when no clustering) it means you have the weighted mean per subject (since the GLM has no condition) in a format IC*freq ; you can use the second level analysis by specify which IC you want per subject (whole brain IC is tricky anyway and would not recommend it at this stage, enter instead a vector of values corresponding to the corresponding row/IC of each subject = perform a test for one cluster of IC)

edwardwl, 2023-01-05:
Of course! Thank you for your help with this. I'll try the strategy that you mentioned.

arnodelorme, 2023-03-24:
Closing the bug for now. Feel free to reopen.

closed 2023-03-24 by arnodelorme