sccn/eeglab #570: Dipfit headmodel

arnodelorme, 2022-12-10:
Use the Fieldtrip default model instead

dipfit/standard_BEM/standard_vol.mat
template = ft_read_headmodel('standard_bem.mat');
template = ft_read_mri('standard_mri.mat');

I indeed believe then to be the same, or at least to come from the same source. It is the BEM model that I made for this simulation study https://doi.org/10.1016/s1388-2457(01)00460-6.

FieldTrip also includes the corresponding electrodes, those are in the fieldtrip/template/electrode directory. The ones that start with "standard_" are the ones that fit on the head. And the MRI is included in fieldtrip/template/anatomy as single_subj_T1.nii (at 2 mm) and single_subj_T1_1mm.nii

Let me do a detailled comparison on the head models...

arnodelorme, 2024-02-09:
```
template = ft_read_headmodel('standard_bem.mat');
```

returns the error

```
Error using ft_notification
file 'standard_bem.mat' does not exist

Error in ft_error (line 39)
  ft_notification(varargin{:});

Error in ft_read_headmodel (line 45)
  ft_error('file ''%s'' does not exist', filename);
```

It only works if I am in the folder xxxx/fieldtrip/template/headmodel
I believe this command used to work anywhere.
@robertoostenveld

robertoostenveld, 2024-02-09:
this is general MATLAB behavior, the file to be loaded should have a full specification of the path, should be in the present working directory, or should be in a directory that is on your path.
 
```
>> template = ft_read_headmodel('standard_bem.mat');
reading 'headmodel' from file 'standard_bem.mat'
% so this works

>> rmpath ~/matlab/fieldtrip/template/headmodel/
>> template = ft_read_headmodel('standard_bem.mat');
Error using ft_read_headmodel
file 'standard_bem.mat' does not exist 
% and this not
```