sccn/eeglab #571: EEGLAB crashes on startup after installing PICARD plugin (issue with pluginstats structure and check version option)

Luke-Acuff, 2022-12-14:
EEGLAB crashes on startup after installing PICARD plugin

This happened after a fresh install of EEGLAB 2022.1 and an attempted install of the PICARD (1.0) plugin. Inspection of the eeglab.m function revealed that the pluginstats structure wasn't generating due to the 'check for updates' option being off. However, pluginstats is called later on anyway, causing the issue. Issue fixed by turning the 'check for updates' option on.

Command Window Log

>> eeglab
Some menus items hidden. Use Preference menu to show them all.
eeglab: options file is ~/eeg_options.m
EEGLAB warning: there can be only one EEGLAB window, closing old one
EEGLAB: adding "ANTeepimport" v1.13 (see >> help eegplugin_eepimport)
EEGLAB: adding "ICLabel" v1.4 (see >> help eegplugin_iclabel)
EEGLAB: adding "PICARD" v1.0 to the pathDot indexing is not supported for variables of this type.

Error in eeglab (line 1054)
                indPlugin = strmatch(lower(pluginlist(plugincount).plugin), lower(pluginstats.name), 'exact');

<br>
#### Steps to Reproduce

1. Download fresh copy of EEGLAB 2022.1
2. Install PICARD (v1.0) via plugin manager

<br>
#### Expected behavior:
EEGLAB runs as usual with new plugin

<br>
#### Actual behavior:
EEGLAB crashes with error listed in above command window output

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Linux RedHat 7.9] |
| Matlab version | [2021a] |
| EEGLAB version | [2022.1] |

arnodelorme, 2022-12-14:
Yes, was able to reproduce the problem. This has been fixed but will only be released. You can replace the [eeglab.m ](https://github.com/sccn/eeglab/blob/develop/eeglab.m) function?

arnodelorme, 2023-03-24:
Closing this report since fixed.

closed 2023-03-24 by arnodelorme