sccn/eeglab #572: Implement parrallel loop for spectopo

neuromechanist, 2022-12-15:
Hi,

Plotting the spectrograms is essential for experimenting with the data, cleaning methods, channel rejection, etc.
A couple of years ago, with rearranging variables on the `spectopo.m`'s main for-loop (~line 930 onward), I could implement a `parfor` to compute the spectrograms for channels/components in parallel.

I wonder if there is an interest in implementing it here as well. The time-saving is ~3x, even for running on a laptop (my example, base model MacBook Pro 14", 2021), and would scale for HD-EEG, longer datasets, and higher core counts.

As an example, for a typical resting-state EEG under the HBN, the main spectopo for-loop takes ~6.5 sec to compute.
Parfor results are slightly varied as the first runs also have an overhead of assigning the memory. It starts with ~3.5 sec for the first run, 2.5 sec for the second run, and 1.5 sec for the third and ongoing runs.

As I plot the spectopo very frequently after each filtering or data cleaning, it is basically 4x the speed.

I also implemented a version with an argument pair `use parallel`, which is `off` by default and can be turned on if the user wants.

Thanks
Yahya

arnodelorme, 2022-12-15:
Yes, a spectrogram plugin would be nice. The MATLAB spectrogram function is pretty powerful.

```matlab
EEG = pop_loadset('filename','eeglab_data.set','filepath','/System/Volumes/Data/data/matlab/eeglab/sample_data/');

[S,F,T,P] = spectrogram(EEG.data(1,:), EEG.srate,[], [], EEG.srate);
figure; imagesc(T, F, abs(S(1:EEG.srate/2,:))); set(gca, 'ydir', 'normal'); caxis([0 1000]);
```

If you overlay events, I think it would look great.

[image]

arnodelorme, 2022-12-15:
Could be your first plugin :-)

neuromechanist, 2022-12-15:
My bad for using the *spectrogram*. I misspoke, I should have said PSD plots, or the spectopo plots such as this one:
```Matlab
pop_spectopo(EEG, 1, [0 EEG.times(end)], 'EEG' ,'percent',100,'freq', [6 10 22], 'freqrange',[2 200],'electrodes','off');
```
[image].

The spectrogram is also very useful; I would love to work on it, actually. But I believe it needs its own issue and discussion.

arnodelorme, 2022-12-15:
I am not sure spectopo would benefit from parfor, mostly because the MATLAB spectral functions is already parallelized. If you do an FFT in MATLAB you will see your processor usage jump to 500% even for a single thread. Also parfor are a pain to handle in a robust way. One function that would be nice to program at the STUDY level is the tftopo function https://eeglab.org/tutorials/11_Scripting/EEG_scalp_measures.html

STUDY already use the parallel toolbox. Let's discuss more what can be done in EEGLAB and what overlaps with your interests.

neuromechanist, 2022-12-15:
Absolutely,
thanks.

closed 2022-12-15 by neuromechanist