sccn/eeglab #574: Custom scripting: "eeg redraw" plus "clear EEG" causes dangerously wrong results (local-global warning)

olafdimigen, 2022-12-16:
Note: Possibly this weird behavior is know or has been fixed by changed Matlab or EEGLAB behavior, but reporting it because effects are extremely severe in my view

### Description
In a very common custom scripting scenario, the EEG is stealthily replaced with that from the previous subject, leading to incorrect results. This happens if `eeg redraw` was executed after loading the data and the variable `EEG` is later manually cleared in the script. It is quite possible that the following is not "orthodox" EEGLAB coding, but I consider this a very common - and therefore dangerous - scenario and it almost got me to wrong results a couple of times

### Steps to Reproduce

Two custom code scenarios, scenario 2 causes problem:

**SCENARIO 1 (everything fine, no clear EEG):** 

```
clear, eeglab % start EEGLAB
for s = 1:2
   % load EEG
   filename = sprintf('myeeg_%i.set',s)
   EEG = pop_loadset(filename)
   eeglab redraw

   % some processing happening here:
  myresult(s) = mean(EEG.data(1,100:200),2);  % get some dependent value
end
```
Result: once subject 2 is loaded, this will give the (custom) warning, that the data in memory (meaning ALLEEG) was overwritten. Nevertheless, this is expected behavior, we get nice results, everybody is happy!

**SCENARIO 2 (problematic behavior with clear EEG):** 
```
clear, eeglab % start EEGLAB
for s = 1:2
   % load EEG
   filename = sprintf('myeeg_%i.set',s)
   EEG = pop_loadset(filename)
   eeglab redraw
   
   % some arbitrary analysis happening here:
   myresult(s) = mean(EEG.data(1,100:200),2);  % get some dependent value

   **clear EEG** % variable EEG is deleted
   % there can be many reasons to clear EEG somewhere mid-script, e.g. because a copy EEG2 was created etc.
end
```
Result: once subject 2 is loaded and redrawn, this will give the dreaded Matlab warning that "The value of local variables may have been changed to match the globals" (apparently happens due to recreating the globals in lines 46-50 of eeg_global.m). However, user is not warned that real mayhem is happening here: entire analysis for subject 2 is actually performed on the EEG of participant 1! 

#### Actual vs Expected behavior:
In scenario 2, due to a simple clear EEG and a hard-to-understand warning, the analysis is computed on wrong data. Expected behavior is that it is run on the newly loaded data of s==2. Also, additionally clearing ALLEEG does not help. I am actually not sure where the old EEG data still lingers around (e.g. if ALLEEG is also cleared)

#### Versions

|   |   |
|----| ---- | 
| OS version     | Win 10 |
| Matlab version | R2021a |
| EEGLAB version | 2021.1 |

arnodelorme, 2022-12-21:
Yes, you are correct. This is annoying, to say the least. Let me think about how to best fix the problem.
I do not think we use the global EEG variables anywhere except the EEGLAB function (and I have checked, maybe 2 other places where it could be removed), so we might just be able to remove global declaration and have eeglab.m access the global workspace variable using 

```
EEG = evalin('base', 'EEG');
```

This could create problems with Octave, so we would need to check carefuly. It is not a small move, but I dislike these global variables as well. 

Probably not for the next release in Jan, not to break anything last minute.

olafdimigen, 2022-12-21:
Great. I believe that this is a rather dangerous one, because it almost "got" me several times despite solid EEGLAB experience and lots of triple-checking, but this is of course a potentially major change. I now finally understood that global variables remain in the global workspace despite a simple `clear` (rather than a `clear global EEG`).

arnodelorme, 2022-12-22:
I have implemented a fix. Not necessarily elegant, but fixes the problem until we remove the global variables. ccec9d067

arnodelorme, 2023-03-24:
There is a new fix in the development version. I have tried your second script and it works.

closed 2023-03-24 by arnodelorme

olafdimigen, 2023-03-24:
excellent, thanks.

arnodelorme, 2023-03-24:
Agreed. It will be dully noted in the next release. It is the first time in 20 years I see it though, so it is specific in how you clear variable etc. While serious, it is also quite rare.
Cheers