sccn/eeglab #576: ICA failed to run on some BIDS datasets

dungscout96, 2023-01-02:
### Description
When running `pop_runica` on ds003465 and ds002718, it failed with error 

"MATLAB:unassignedOutputs
Output argument "sphere" (and maybe others) not assigned during call to "runica"."
<br>
#### Steps to Reproduce
ALLEEG = pop_runica(ALLEEG, 'icatype','runica','concatcond','on','options',{'pca',-1, 'extended', 'on', 'lrate', lrate, 'maxsteps', 2000});

<br>

#### Expected behavior:
ICA run successfully

<br>

#### Actual behavior:
Failed with error MATLAB:unassignedOutputs

<br>

#### Versions

|   |   |
|----| ---- | 
| OS version     | Lab Computing cluster|
| Matlab version | MATLAB version on computing |
| EEGLAB version | latest changes from develop |

arnodelorme, 2023-03-24:
Can you provide the exact dataset?

arnodelorme, 2023-05-12:
@dungscout96  exact dataset to try on?

arnodelorme, 2023-05-25:
ds002718 is all good in the pipeline and we do not have ds003465 anymore. Closing this.

closed 2023-05-25 by arnodelorme