sccn/eeglab #577: pop_resample error at line 224 for events with latency <1 sec

alenarto, 2023-01-04:
### Description

Running pop_resample(EEG,200) produces the error:

_Array indices must be positive integers or logical values.

Error in pop_resample (line 224)
                EEG.event(iEvt).latency = (EEG.event(iEvt).latency - bounds(iBnd)) * p / q + indices(iBnd);_

I believe this is because the previous line: 
(223) iBnd = sum(EEG.event(iEvt).latency >= bounds); 

results in a value of 0 for iBnd, which then creates the error above when trying to index bounds(0). 

This problem arises because I have events with a latency less than 1, and the bounds are defined as [1 x]. The >= test then produces a 0 when matching a value less than 1 agains [1 x]. When I change the bounds to [0 x], the code completes the updating of the event latencies.

The sampling rate is not the issue here as I tried this with other sampling rates and on other files and had no issues. It errors out only when events exist with latency <1 that are not boundary events.

<br>
#### Steps to Reproduce

1. Run pop_resample(EEG, 200) with data with events with latency <1.

<br>
#### Expected behavior:
Resample completes.

<br>
#### Actual behavior:
The resampling completed, but the function failed in updating event latencies.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Linux Cluster HPC Centos7] |
| Matlab version | [r2020a] |
| EEGLAB version | [v2022.0] |

arnodelorme, 2023-01-04:
Let me look into that.

arnodelorme, 2023-03-24:
Fixed in 0ac069bab

closed 2023-03-24 by arnodelorme