sccn/eeglab #578: study creation corrupts set file (std_editset, eeg_store, eeg_getdataact)

alenarto, 2023-01-06:
### Description

The process of creating a study (specifically via GUI "browse for datasets" modifies the files), makes them unreadable in subsequent routines, such as trying to precompute a measure, or even loading files. I observed this in v2022 and v2022.1. 

I struggled to figure out what's going on but when I compared an EEG set file pre vs post study creation process I noted that post study creation the EEG.data field says "in set file". I then manually changed this field to read "filename.fdt" so that the code has the actual name of the filename for the data field. I used Matlab's save command to save this EEG struct as the "filename.set" file and when I attempted to load this set file via pop_loadset, it now worked fine.

So my best guess is that the study creation process tries to replace the EEG.data field with a pointer to the data ("in set file") that can not be decoded correctly by later scripts (eeg_getdatact.m). I confirmed this by creating  the study with the same datasets but with the "update data sets" check box set to OFF, so that study and data info is kept separate.

<br>
#### Steps to Reproduce
Through GUI:
1. Select create study > browse for datasets.
2. Select two random datasets, add subject ID. Leave defaults (with update datasets "on") and press ok.
3. Save study as.
4. Try to load any of the study datasets (e.g., select dataset through gui or try to precompute channel features) - original attempt was to try to precompute power spectra using all defaults.

<br>
#### Expected behavior:
I expect the precompute to precompute power spectra for chosen files.

<br>
#### Actual behavior:
The code hangs. There is zero output and Matlab says it's "busy". Digging into this I found that the code is stuck in eeg_getdataact (in std_spec),  with cntrl-c sometimes spitting me back out from eeg_store.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Centos7 on HPC] |
| Matlab version | [2020a] |
| EEGLAB version | [2022.0 and 2022.1] |

arnodelorme, 2023-03-04:
I think this has been fixed. Would you mind to double check?

arnodelorme, 2023-03-24:
@alenarto can you check if it is fixed for you?

alenarto, 2023-04-01:
Hi @arnodelorme - sorry I missed this. Which version should I check?

arnodelorme, 2023-04-02:
The latest version from Github

alenarto, 2023-04-03:
@arnodelorme using version EEGLAB dev I can confirm that I could create a study and precomputed files. Did not see EEG.data field says "in set file" behavior as previously. The data field (when loaded manually) instead shows the file name: e.g., EEG.data: 'sub-21002_task-cogs_eeg1.fdt'
Agatha

arnodelorme, 2023-04-03:
Wonderful. Thanks Agatha. Keep on the good bug reports going, and always happy to jump on a call if you are stuck to fix a problem.

closed 2023-04-03 by arnodelorme

arnodelorme, 2023-04-04:
Thanks Agatha