sccn/eeglab #582: pop_clean_rawdata channel cleaning unexpected behavior

alenarto, 2023-01-21:
### Description

There's a bug of sorts in clean_artifacts.m when the user requests channel cleaning (without any other ASR options), with the inclusion of several channels for omission. So let's assume the following command (which I describe via GUI below - both GUI and command line produce same behavior): 

EEG = pop_clean_rawdata(EEG, 'FlatlineCriterion',5,'ChannelCriterion',0.8,'LineNoiseCriterion',4,'Highpass','off','BurstCriterion','off','WindowCriterion','off','BurstRejection','off','Distance','Euclidian','channels_ignore',{'M1','M2'});

I am requesting ONLY channel cleaning (no ASR), with default settings PLUS exclusion of M1 and M2, which are still in the channel set. This command eventually calls clean_artifacts.m which at line 252 cleans the channels via:

[EEG,removed_channels] = clean_channels(EEG,chancorr_crit,line_crit,[],channel_crit_maxbad_time, num_samples); 

Up to here everything works fine. The output EEG struct has n-x channels, with x being the number of removed_channels. However then at line 296 the code triggers an if statement because there are omitted channels, that attempts to add back in the original omitted channels into the EEG struct. However this code actually also adds back in the "removed_channels", along with the "omitted channels". So the output of this entire routine is the original data. Essentially it overwrites the clean_channels output.

FYI - if I exclude the "omit channels" option it does what it's supposed to. Namely it produces an EEG struct with the bad channels omitted (excluded).

Note. The behavior of this routine seems clunky for channel cleaning WITHOUT ASR. It seems like it should have an option to interpolate the bad channels, or just indicate which are bad without modifying the struct. As is, it runs pop_select to remove those channels without giving the user a choice how to correct.

<br>
#### Steps to Reproduce

1. Via Gui, launch ASR-cleaning menu (pop_clean_rawdata interface).
2. Turn off (uncheck) everything except the channel cleaning options.
3. Select all defaults, but in addition select the "omit" channel button and pick some channels to omit.
4. Run.

<br>
#### Expected behavior:
I expect to get an EEG struct with dirty channels now cleaned, and my omitted channels untouched.

<br>
#### Actual behavior:
I get original data despite report from code that 3 channels were cleaned.

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [Centos7 on HPC] |
| Matlab version | [2020a] |
| EEGLAB version | [2022.1] |

arnodelorme, 2023-03-04:
This is a good point. We will look into it. Apologies for the late reply.

clementlee2000, 2023-03-15:
Good to see you're still writing such thorough bug reports Agatha! Hope you are well (and that this gets resolved soon) 😄

arnodelorme, 2023-03-15:
Agatha,

Which version are you using? 

I cannot find the code "However then at line 296 [of clean_artifacts.m] the code triggers an if statement because there are omitted channels, that attempts to add back in the original omitted channels into the EEG struct. However this code actually also adds back in the "removed_channels", along with the "omitted channels". So the output of this entire routine is the original data."

arnodelorme, 2023-03-15:
OK, I think I got it. It is below.

arnodelorme, 2023-03-15:
Fixed in 836ce44
Let me know if it works for you.

closed 2023-03-15 by arnodelorme