sccn/eeglab #583: EEGLAB older version 2022.* not available for download

DimitriPapadopoulos, 2023-01-24:
### Description

According to the [EEGLAB revision history](https://eeglab.org/others/EEGLAB_revision_history.html), EEGLAB releases include:
* [EEGLAB version 2022.1](https://eeglab.org/others/EEGLAB_revision_history.html#eeglab-version-20221)
* [EEGLAB version 2022.0](https://eeglab.org/others/EEGLAB_revision_history.html#eeglab-version-20220)
* [EEGLAB version 2021.1](https://eeglab.org/others/EEGLAB_revision_history.html#eeglab-version-20211)
* [EEGLAB version 2021.0](https://eeglab.org/others/EEGLAB_revision_history.html#eeglab-version-20210)

According to [Download EEGLAB](https://sccn.ucsd.edu/eeglab/downloadtoolbox.php):
> EEGLAB is released twice a year and the version name is based on the year. EEGLAB older versions are available [here](https://sccn.ucsd.edu/eeglab/download/daily/).

Yet https://sccn.ucsd.edu/eeglab/download/daily/ does not include older version **eeglab2022.0.zip** or **eeglab2022.1.zip** - but I guess  the latter is available as https://sccn.ucsd.edu/eeglab/currentversion/eeglab_current.zip instead.

#### Steps to Reproduce

1. Go to https://sccn.ucsd.edu/eeglab/download/daily/.
2. Look at the available files
3.  Notice that **eeglab2022.0.zip** is missing.

arnodelorme, 2023-03-04:
Good point. This is an issue with our HTTP vs FTP access. I have placed the missing files where they belong.

closed 2023-03-04 by arnodelorme

DimitriPapadopoulos, 2023-03-04:
Thank you @arnodelorme. I guess **2023.0** is currently available as `eeglab_current.zip`, isn't it?

arnodelorme, 2023-03-04:
correct