sccn/eeglab #586: EEGLAB error in function runica() at line 873

xiaoya675, 2023-02-16:
### D
[image]
escription

[*Description of the bug or feature
Attach a link to Dropbox or Google Drive with screenshots and your data if possible.*]

<br>
#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

arnodelorme, 2023-03-04:
Would you mind sharing a dataset so we can test?

arnodelorme, 2023-03-24:
Closing this but feel free to reopen.

closed 2023-03-24 by arnodelorme