sccn/eeglab #587: Biosig not installing on Macbooks - missing eeglab files?

AndiMcGrunt, 2023-02-16:
[image]
We have been having problems getting Biosig installed on Macbooks - two different machines at this point. Both are fresh, new installs of Matlab (v. R2022b) and the full EEGlab version (not from github) of 2020.0 (28.8 MB). When we try to import edf files using Biosig for the first time in eeglab, we cannot autoinstall Biosig as has worked on my work PC. We have attempted to manually install Biosig with experienced Matlab users from our university, and we continue to get this error message. They suggest there may be missing files from our version of eeglab. Thanks for any ideas you have!

arnodelorme, 2023-03-04:
This is an issue with the firewall on your side. You can always download the plugins manually and unzip them in the eeglab/plugins folder. https://sccn.ucsd.edu/eeglab/plugin_uploader/plugin_list_all.php

closed 2023-03-04 by arnodelorme

AndiMcGrunt, 2023-03-04:
Thank you - as I said, we have tried downloading the biosig plugin and installing it manually on both MacBooks. This has not worked.

arnodelorme, 2023-03-04:
You need to place "biosig4matlab" in the plugin folder, not the entire biosig uncompressed folder. Let me know.

AndiMcGrunt, 2023-03-15:
Hi Arnaud – Thank you, we did attempt a manual install before, but that didn’t work. We’ve had some time to work with it this week and seems the issue was resolved with the latest version of EEGlab, biosig installed fine with that version and everything seems to be working fine now.  Appreciate your help!

Andi