sccn/eeglab #588: eeg_interp.m line271 bug

sgeirnae, 2023-02-21:
### Description

Line 271 of eeg_intepr.m defines tmpdata as a tensor where the first dimension correspond to the number of bad channels. On line 272, the originally good channels are restored. On line 289, the interpolated bad channels are assigned. Implicitly, these means that in case you have bad channels without coordinates (for example, when you have custom channels), they are replaced by zeros. 

In case your last channel is such a custom channel that is bad, this routine goes wrong, as this will cause dropping this last channel. This causes a mismatch between the number of channels and labels, which will cause eeg_checkset on line 293 to drop all channel information.

I don't think this behaviour is wanted or desirable - this last bad channel should still be present (even when replaced with zeros). Problem seems to be with the initialization on line 271 - it seems to make much more sense to initialize with the number of original channels (instead of bad channels).

arnodelorme, 2023-03-04:
We implemented a fix last week. Would you mind testing again?

sgeirnae, 2023-03-06:
In the eeglab2023.0 release (if the update is included in this version), the problem still persists - if the last channel is indicated as bad while no coordinate info is available, it is still removed.

arnodelorme, 2023-03-08:
Would you mind sharing a dataset that has that problem and steps to reproduce the problem (which menu to call or command to use on the MATLAB prompt).

sgeirnae, 2023-03-14:
I noticed it's not only with the last channel. Any channel without coordinates (could happen if you have custom electrodes) will be removed, while the channel information is not updated.

1. Import an EEG dataset with channel information. Assume there are N channels.
2. Pick a random channel and remove the channel information (here: channel 1): EEG.chanlocs(1).theta = []; EEG.chanlocs(1).radius = [];  EEG.chanlocs(1).sph_theta = [];  EEG.chanlocs(1).sph_phi = [];  EEG.chanlocs(1).sph_radius = []; EEG.chanlocs(1).X = [];  EEG.chanlocs(1).Y = [];  EEG.chanlocs(1).Z = [];
3. EEG_intp = eeg_interp(EEG,1);

Now this first channel will be removed in the output, i.e., EEG_intp.nbchan == N-1 and size(EEG_intp.data,1) == N-1, while length(EEG_i.chanlocs) == N. This inconsistency makes it hard to process afterwards. I think at least this inconsistency should be fixed? Whether eeg_interp should remove a bad channel when there is no coordinate info is a design choice (I would prefer keeping that channel).

arnodelorme, 2023-03-14:
That's interesting. But you are trying to interpolate a channel that does not have a location, so it is impossible. The only thing we could do is ignore this channel and not interpolate it. Is that what you expect?

sgeirnae, 2023-03-15:
Yes, I agree that it is impossible, but it sometimes happens in the chain of your commands (first detecting bad channels etc.). Being aware of the issue is the responsibility of the user, but eeg_interp could help flag the issue?

Ideally, I would expect indeed ignoring the channel and not interpolating + throwing a warning that it cannot be interpolated - then the user can decide whether to throw it out or not. At the very least, the output should be consistent (i.e., chanlocs and the data).

arnodelorme, 2023-03-24:
I am not sure it is a good solution. Say you are trying to interpolate 10 channels, but one is empty, then it would only interpolate 9, and the size of your resulting dataset would not be what you expect.

I have added a custom error.

```
Error using eeg_interp
Trying to interpolate a channel without coordinates assigned to it
```

closed 2023-03-24 by arnodelorme