sccn/eeglab #596: Error in function ftestc() undefined function "finv" for input "doubles"

estebandleonc, 2023-03-18:
### Description

Hello Arnauld,

I have been using the cleanline plugin without any issues and suddenly today I can't use it anymore. It was sending me a problem in funtion dpsschk():
Undefined funtion 'dpss' for input arguments of type 'double'. And I found in github that a person had a similar problem (#153) and you answered: "Cleanline only works with the Matlab signal processing toolbox (an additional paying toolbox) and apparently you are missing that function. I have added a note to the cleanline repository".

I erased matlab and eeglab, I downloaded them again and downloaded the signal processing toolbox (I didn't pay, I think my university has it). 

Now it sends me this: Error in function ftestc(): undefined function "finv" for input arguments of type "doubles".  Moreover, I can't visualise anymore the scrolling data window with the rejected data highlighted and I receive this message: "vis_artifacts failed. Skipping visualization. Could be because of duplicate channel label". 

As mentioned before, everything was working perfectly, I haven't done anything different and I didn't need any new toolboxes to run every single step of the pipeline (I read somewhere that I need now the statistics and machine learning toolbox). The first step in the cleanline and the second one is the rejection of bad channels so please if you can help me telling me what to do would be amazing. I am pretty new at this.

The path has been set up as always and this is the message that appears before opening (in version 2022.1):
eeglab: options file is C:\Users\Esteban\eeg_options.m
Retrieving plugin versions from server...
Retrieving download statistics...
EEGLAB: adding "Cleanline" v2.00 (see >> help eegplugin_cleanline)
EEGLAB: adding "ICLabel" v1.4 (see >> help eegplugin_iclabel)
EEGLAB: adding "bva-io" v1.71 (see >> help eegplugin_bva_io)
EEGLAB: adding "clean_rawdata" v2.7 (see >> help eegplugin_clean_rawdata) - new version 2.8 available
EEGLAB: adding "dipfit" v4.3 (see >> help eegplugin_dipfit) - new version 5.1 available
EEGLAB: adding "firfilt" v2.5.1 (see >> help eegplugin_firfilt) - new version 2.7.1 available

I am using the same EEG file with the same pre-processing steps so the problem is not the file.

| OS version     | [windows] |
| Matlab version | [2023a] |
| EEGLAB version | [2022.1 and 2023] |

Thank you very much for your help,

Esteban Leon

arnodelorme, 2023-03-24:
Hi Esteban, this function finv is from the statistics toolbox. Can you download it and check that it works?

arnodelorme, 2023-03-24:
Closing this bug but feel free to reopen.

closed 2023-03-24 by arnodelorme

estebandleonc, 2023-03-25:
Hello Arnaud, yes it worked. Apparently previously it was downloaded automatically with the signal processing toolbox and the statistics toolbox, and this time it only downloaded it with the signal processing toolbox. Everything is working fine. Thank you so much!

arnodelorme, 2023-03-25:
Good to know