sccn/eeglab #598: Manage Extensions: Plugin Error

kvanbenthem, 2023-03-26:
### Error with Manage Extensions

[*I have a new error that occurs when I try to go to the Manage Extensions in the main EEGLAB GUI*]
[image]

<br>
#### Steps to Reproduce

1. [*Open EEGLAB*]
2. [*Go to File Tab*]
3. [*Click on Manage Extension*]

<br>
#### Expected behavior:
[*I should get a GUI to add or update the extensions and plugins*]

<br>
#### Actual behavior:
[*Unrecognized field name plugin  in function plugin_getweb() line 89  *]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 10*] |
| Matlab version | [* 2023a current version*] |
| EEGLAB version | [*v2023.0 current version*] |

kvanbenthem, 2023-03-28:
The error message is this: “Unrecognized field name “plugin” in function plugin_getweb() line 89 “

kvanbenthem, 2023-03-28:
In the command window of Matlab the last line is this: "Retrieving download statistics..."

arnodelorme, 2023-04-08:
I think it is because it fails to connect to the internet -- even though the internet is on. I have added an additional check in 0f2226c67

closed 2023-04-08 by arnodelorme