sccn/eeglab #601: EEG.chaninfo.removedchans field error?

stellafychang, 2023-04-01:
### Description

I have been using EEG.chaninfo.removedchans for monitoring the removed channels after ASR. After I updated MATLAB and EEGLAB version, I found that the structure of EEG.chaninfo.removedchan has become different. After ASR, I will interpolate the removed channels and I found that the removed channels were still interpolated correctly, so I'm not sure if the information of the removed channels has been stored in a different location now.  

<br>
#### Steps to Reproduce

1. *First Step*: run ASR and get some channels to be rejected
2. *Second Step*: check the content of EEG.chaninfo.removedchans
<br>
#### Expected behavior:
<br>
The removed channels information will be in EEG.chaninfo.removedchans.
<br>

[image]

<br>
#### Actual behavior:
<br>
I was supposed to see 40 removed channels, but they cannot be found.
<br>

[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11*] |
| Matlab version | [*R2023a*] |
| EEGLAB version | [*2023.0*] |

arnodelorme, 2023-04-04:
Let me check. The information of the channels removed by ASR is also in EEG.etc.clean_channel_mask

arnodelorme, 2023-04-08:
I have removed channels in 2 passes of clean_rawdata and both were saved appropriately (the second pass adding to the first pass). I think this bug refers to an older version of clean_rawdata. 

If you think there is still a problem with the latest version of clean_rawdata, let us know (and upload data so we can test). Thank you so much.

closed 2023-04-08 by arnodelorme