sccn/eeglab #605: EEGLAB error in function dipplot() at line 691

stellafychang, 2023-04-06:
### Description

I tried to plot the dipoles using GUI at STUDY level, but there was a pop-up window showed up: EEGLAB error in function dipplot() at line 691.

<br>
#### Steps to Reproduce

1. *First Step:* please load the STUDY from the Issue 516
2. *Second Step:* go to 'STUDY' -> 'Edit/plot components clusters'

<br>
#### Expected behavior: I could get the plot of dipoles with EEGLAB 2021.1

<br>

[image]

<br>
#### Actual behavior: After switching to EEGLAB2023.0, I got this error.

[image]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | MacOS Ventura 13.2.1 |
| Matlab version | R2023a |
| EEGLAB version | 2023.0 |

arnodelorme, 2023-04-08:
We have updated Dipfit. Can you update your version in the plugin manager and check that it works?

stellafychang, 2023-04-11:
Yes. I updated Dipfit to 5.2 and it works now. Thank you.

closed 2023-04-11 by stellafychang

arnodelorme, 2023-04-11:
Wonderful. Thanks