sccn/eeglab #607: pop_clean_rawdata error

dungscout96, 2023-04-06:
### Description
Ran into error when running pop_clean_rawdata on attached dataset. Potentially because data too short. Arguments used:
```
options = {'FlatlineCriterion',5,'ChannelCriterion',0.85, ...
                    'LineNoiseCriterion',4,'Highpass',[0.75 1.25] ,'BurstCriterion',50, ...
                    'WindowCriterion',0.25,'BurstRejection','on','Distance','Euclidian', ...
                    'WindowCriterionTolerances',[-Inf 7] ,'fusechanrej',1};
```

<br>
[sub-020_task-AuditoryVisualShift_run-01_eeg.set.zip](https://github.com/sccn/eeglab/files/11173467/sub-020_task-AuditoryVisualShift_run-01_eeg.set.zip)

#### Steps to Reproduce

1. [*First Step*]
2. [*Second Step*]
3. [*and so on...*]

<br>
#### Expected behavior:
[*What you expected to happen*]

<br>
#### Actual behavior:
[*What actually happened*]

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*fill in*] |
| Matlab version | [*fill in*] |
| EEGLAB version | [*fill in*] |

dungscout96, 2023-04-06:
[sub-020_task-AuditoryVisualShift_run-01_eeg.set.zip](https://github.com/sccn/eeglab/files/11173581/sub-020_task-AuditoryVisualShift_run-01_eeg.set.zip)

arnodelorme, 2023-04-08:
OK, it is now outputing

```
No enough reference data, call the function from the command line and try changing the 'BurstCriterionRefTolerances' parameters
```

closed 2023-04-08 by arnodelorme