sccn/eeglab #609: Issue with channel lookup

arnodelorme, 2023-04-13:
Load up this file

EEG = pop_biosig('/Users/arno/Downloads/ds003343/sub-13/ses-01/eeg/sub-13_ses-01_task-fps_run-01_eeg.bdf');

- And then try channel lookup from the GUI
- Then scroll to next channel and there is an error

sdebener, 2023-04-20:
Hi Arno,
we have a very similar/related issue, pointing to line 1033 in pop_chanedit (see below). My students are frustrated, they cannot use eegh anymore to add channel locations with pop_chanedit ;-/. We use EEGLAB2023.0 in class....
Cheers,
Stefan

*******
IMPORTANT: After importing/modifying data channels, you must close
the channel editing window for the changes to take effect in EEGLAB.
TIP: Call this function directly from the prompt, ">> pop_chanedit([]);"
     to convert between channel location file formats
readlocs(): 'loc' format assumed from file extension
31 channel with the same label found and imported
Channel lookup: no location for Nz,LPA,RPABrace indexing is not supported for variables of this type.

Error in pop_chanedit (line 1033)
    userdata.commands = { userdata.commands{:} args{:} };
 
Error while evaluating UIControl Callback.

Brace indexing is not supported for variables of this type.

Error in pop_chanedit (line 1033)
    userdata.commands = { userdata.commands{:} args{:} };
 
Error while evaluating UIControl Callback.

arnodelorme, 2023-04-20:
Closing this for now, but please reopen if you encounter any problem.

closed 2023-04-20 by arnodelorme