sccn/eeglab #612: Error with ICLabel

rohitgarud, 2023-04-27:
Hello,
I am new to EEGLab. I am getting the following error when I try to Label the ICA components using ICLabel.
```
Error using dagnn_bc.Conv
Error: File: Conv.m Line: 1 Column: 10
Class name and filename must match.

Error in dagnn_bc.DagNN.loadobj (line 26)
    block = constr() ;

Error in run_ICL (line 30)
    net = dagnn_bc.DagNN.loadobj(netStruct);

Error in iclabel (line 56)
labels = run_ICL(version, features{:});
```

I am using MATLAB 2020b and EEGLab version 2023, ICLabel version 1.4. I have tried installing older versions of ICLabel but same error is shown. Please help me solve this issue.

Thank you

arnodelorme, 2023-04-27:
Which version of ICA label are you using? Can you check that the folder 

eeglab/plugins/ICLabel/matconvnet

is present and not empty

rohitgarud, 2023-04-28:
Thank you @arnodelorme for your prompt response. 
I am using ICLabel version 1.4. I have installed it using the Extension manager from EEGLAB and checked that the folder is not empty.

rohitgarud, 2023-04-28:
When I install EEGLAB with MATLAB runtime, it works. So I think the issue is with the MATLAB installation.

arnodelorme, 2023-04-28:
Ah good to know.

closed 2023-04-28 by arnodelorme