sccn/eeglab #614: Error when adding old ref. channel back to the data in v2023.0

QingkaiZhu, 2023-05-03:
### Description

I want to retain the old reference channel and then compute the average reference but got an error in the pop_reref window.
[image]

This is a regression bug, it works in eeglab v2022.1 and failed in v2023.0

### Steps to Reproduce

Just follow the steps in the official tutorial with _eeglab_data.set_
- https://eeglab.org/tutorials/05_Preprocess/rereferencing.html#retaining-the-reference-channel

#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Windows 11*] |
| Matlab version | [*R2022b*] |
| EEGLAB version | [*v2023.0*] |

arnodelorme, 2023-05-12:
Did you add the old reference to the list of channels as recommended in the tutorial?

arnodelorme, 2023-05-25:
I tried the tutorial and it worked for me. I am closing this report, but feel free to reopen.

closed 2023-05-25 by arnodelorme

carrie372, 2024-02-07:
@arnodelorme when I follow the tutorial I end up with two Cz channels. One for channel 24 and the new one appended in channel locations.