sccn/eeglab #615: Cross-Coherence

MarioBert, 2023-05-03:
I am working on event-related coherence in some Zeto data and came across UCSD powerpoints showing it could be done via Plot --> Time-frequency transformations --> Channel or component cross-coherence 

These powerpoints are from 2017 and on my version of EEG lab the "Time-frequency transformations" is not an option on the dropdown menus, so I attempted to call it using pop_newcrossf() but receive the following error:

Error in pop_newcrossf (line 186)
pointrange1 = round(max((tlimits(1)/1000-EEG.xmin)*EEG.srate, 1));

It seems like there is something wrong with my epoch time limits but I have tried many different combinations and have received the same error.
 
Any chance you have seen this before or could point me in the direction of someone that has?

arnodelorme, 2023-05-12:
We would need to see the data and the call that is being used. Any chance you can share that?

arnodelorme, 2023-05-25:
Closing this for now, but feel free to send the data and we will have a look.

closed 2023-05-25 by arnodelorme