sccn/eeglab #616: EDF format bug

KseniiaKriukova, 2023-05-09:
Hello, 

When trying to plot the EEG data (edf format) the following mistake occurs 
[image]

At the same time EEGLab shows the following warning: 
Warning: EDFANNOT2EVT.M is currently not supported! If there are problems reading EDF files, 
> In edfannot2evt (line 26)
In sopen (line 1056)
In pop_biosig>readfile (line 324)
In pop_biosig (line 207) 
Warning: download and install mexSLOAD.mex* from https://pub.ist.ac.at/~schloegl/src/mexbiosig/ 
> In edfannot2evt (line 27)
In sopen (line 1056)
In pop_biosig>readfile (line 324)
In pop_biosig (line 207) 
____________
https://pub.ist.ac.at/~schloegl/src/mexbiosig/  - the following website seems to be unavailable. [Screenshot 2023-05-09 at 2.36.42 PM.png (604K)](https://mail.google.com/mail/u/0?ui=2&ik=6dae198799&attid=0.2&permmsgid=msg-a:r2474830249455652283&view=att&disp=safe&realattid=f_lhgsl1in1) 
[image]

May I kindly ask you how this issue might be fixed?

Thanks a lot in advance.  

Kind regards

arnodelorme, 2023-05-09:
Can you share you EDF data file?

KseniiaKriukova, 2023-05-10:
Thank you so much!
I have clinical data and unfortunately am not allowed to share it, at the
same time I used to work with the same data in the EEGLab before and
everything was okay.
Thank you again

arnodelorme, 2023-05-12:
Unfortunately, we cannot fix the issue without seeing the file. Perhaps, you can share an anonymized version of the file?

closed 2023-05-12 by arnodelorme