sccn/eeglab #618: clean_channels misses channels with large amplitude?

amisepa, 2023-05-11:
Trying to create bad channels in a file for a tuto (by simply multiplying the signal by 3 for FC3 and by 15 for AF8), to show them in an illustration with vis_artifacts. To my surprise, clean_channels did not detect them despite huge difference in amplitude, and removes other channels that look good. There must be some rescaling in vis_artifacts because my bad channels are not visible in vis_artifacts. But they are still there. 

Channels that should be removed: FC3 and AF8
[image]

Channels removed instead by vis_artifacts: T7 PO7 O1 --> notice how the bad channels are rescaled and not visible here. 
[image]

Plot after clean_rawdata shows that FC3 and AF8 are still in the data. 
[image]

STEPS:

Load "sample_data2.2.set" from here: https://github.com/amisepa/BrainBeats/tree/main/sample_data

    oriEEG = EEG;
    pop_eegplot(EEG,1,1,1);   % here you can see FC3 and AF8 are clearly bad
    EEG = pop_clean_rawdata(EEG,'FlatlineCriterion',5,'ChannelCriterion',.85, ...
        'LineNoiseCriterion',5,'Highpass','off', 'BurstCriterion','off', ...
        'WindowCriterion','off','BurstRejection','off','Distance','off');    
        vis_artifacts(EEG,oriEEG);   % clean_channels removed 
    badChan = ~contains({oriEEG.chanlocs.labels}, {EEG.chanlocs.labels});
    fprintf(1, 'Bad channels were: ');
    fprintf(1, '%s ', oriEEG.chanlocs(badChan).labels )
    fprintf(1, '\n')
    pop_eegplot(EEG,1,1,1);

arnodelorme, 2023-05-12:
This is because clean_rawdata uses the correlation between channels, and you did not destroy the correlation by multiplying them. So it is all good. Your artifacts are not physiologically plausible.

closed 2023-05-12 by arnodelorme

amisepa, 2023-05-12:
Ok thanks! would it work if I reverse polarity? or do I need to shuffle data to create a fake bad channel?

arnodelorme, 2023-05-12:
Reversing polarity will not work either. You can take the mirror time image.

EEG.data(1,:) = EEG.data(1:end:-1:1)

amisepa, 2023-05-12:
Great thanks