sccn/eeglab #619: clean_channels help is incorrect

amisepa, 2023-05-12:
function [signal,removed_channels] = clean_channels(signal,corr_threshold,noise_threshold,window_len,max_broken_time,num_samples,subset_size)

% Signal = clean_channels(Signal,CorrelationThreshold,WindowLength,MaxBrokenTime,NumSamples,SubsetSize,UseGPU)

--> missing noise threshold and GPU should be subset_size in help

arnodelorme, 2023-05-12:
I don’t get it.

amisepa, 2023-05-12:
function: clean_channels(signal, corr_threshold, noise_threshold, window_len, max_broken_time, num_samples, subset_size)

help: clean_channels(Signal, CorrelationThreshold, WindowLength, MaxBrokenTime, NumSamples, SubsetSize, UseGPU)

3rd input of help should be noise_threshold and last should be subset size (not UseGPU)

arnodelorme, 2023-05-12:
Thanks Done.

closed 2023-05-12 by arnodelorme