sccn/eeglab #626: assert error

evastone97, 2023-05-23:
Hello I had this warning after i done ICA , I tried to label my components and I got this assert error message. What might have been wrong?

[image]

arnodelorme, 2023-05-25:
Your ICA data has complex number. How did you compute ICA?

evastone97, 2023-05-26:
Hello, 

I compute ICA with the runica algorithm and I did a dimensionality reduction of 30. I have a screenshot below:
[image]

Note: the data have been preprocessed (single files for each subject) with ica and then IC Label was used on them for artifact rejection.
Is the error created because of double ICA , or because the data before have been preproceeced with ica and ic label in single files? ( ica should be in the whole study probably from the beggining so also for artifact rejection and then I should continue from there?)

what I was trying to do afterwards is to create a study including all the single files of the subjects, implement dimensionallity reduction and then classify (but I am stack in IC label)

arnodelorme, 2024-02-09:
Closing this issue for now, but let me know if the issue persist.

closed 2024-02-09 by arnodelorme