sccn/eeglab #629: Unable to run coregister.m when using its align fiducials feature.

leima1, 2023-06-08:
### Description

Unable to run 'coregister.m' when using its align fiducials feature.

In line 353 of `coregister.m`: `[ electransf transform ] = align_fiducials(electmp, elec2, g.alignfid);`
the align_fiducials function is given three inputs. 

However, the nested align_fiducials function in line 637 asks for four inputs  
`function [elec1, transf] = align_fiducials(elec1, elec2, fidnames1, fidnames2)`

Errors
1. `electmp` is never defined in the function, giving an error when running line 353. Perhaps `electmp` is meant to be `elec1`?
2. align_fiducials is expecting a 4th input but it is never given. Perhaps fidname2 is not supposed to be passed in? Alternatively, lines 641 to 643, `elec1.label{ind1} = fidnames2{1}; ...` is meant to be written as `fidnames2{1} = elec1.label{ind1}` and so on?

Making those changes appears to fix the errors, and I can get the transform array. But would like to double-check to make sure the changes are correct. Thank you! 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | [*Mricrosoft Windows 10 Enterprise*] |
| Matlab version | [*2022b*] |
| EEGLAB version | [*the latest version on GitHub*] |

arnodelorme, 2023-06-26:
Yes you are correct. This was fixed in eeda38f55 
This is an issue we have not encountered before. It would be great if you could test the function in the same way you were testing it before and let us know if it works as expected.

leima1, 2023-06-27:
Thanks, Arnaud. One of the error still exist:

```
Not enough input arguments.

Error in coregister>align_fiducials (line 641)
    ind1 = strmatch(fidnames1{1}, elec1.label, 'exact'); elec1.label{ind1} = fidnames2{1};

Error in coregister (line 353)
        [ ~, transform ] = align_fiducials(elec1, elec2, g.alignfid);
```

The nested function` [elec1, transf] = align_fiducials(elec1, elec2, fidnames1, fidnames2)` starting on line 637 is expecting 'fidnames2' However, that is not passed in when the function is called on line 353.

arnodelorme, 2023-06-27:
I think this option is no longer used from the GUI. Instead now it is using rigid warping for fiducials.

leima1, 2023-06-27:
Yeah, the GUI works fine for me. 

I was calling `coregister` in my custom script with `alignfid` as one of the input options to align the fiducials, and these errors came out. I did not realize it was obsolete. I am now using the `warp` option instead, and it appears to work just fine.

arnodelorme, 2024-02-09:
Changed the help message.

closed 2024-02-09 by arnodelorme