sccn/eeglab #635: resampling data

droe11, 2023-06-19:
How to resample data in eeglab once ICA and epoch have been performed?

I am trying to resample my EEG data to reduce computation time for MVPA analysis, the data was filtered at 512Hz and I am now trying to downsample to 256Hz. 

The data has been preprocessed, epoched and ICA done on the data. 

When I try and resample, either through eeglab or MATLAB code, I am presented with the following error message: "Dimensions of arrays being concatenated are not consistent., (Error occurred in function resample() at line 122). 

I am not sure which dimensions of the array this error message is referring to or how to concatenate these together, any advice would be greatly appreciated! 

[image]

I expected to be able to just follow the GUI process and downsample the data, does this need to be done before any other preprocessing steps have taken place? 

I am using MATLAB version 2022a, EEGLAB v14.1.2. 

Thank you

gbShaw, 2023-06-22:
> I got the same error message, and don't know how to solve this problem.
> 
> My data is continous (not epoched), the error message is same

I know how to solve my problem,  it is the fieldtrip's function in matlab search path, conflicted with eeglab's  resample.m function.
after remove fieldtrip from search path, every thing goes well

droe11, 2023-06-22:
Hi Zhibing,

Thank you !!!

Removing the field trip lite folder that comes with eeglab from the matlab
path made it work straight away!

robertoostenveld, 2023-06-26:
It seems to be in https://github.com/fieldtrip/fieldtrip/blob/bfae6ab269176a5fc796090b7e660d6763f4efeb/external/signal/resample.m#L122, which is not a fieldtrip function, but an octave function in a directory that users can add to the path (for fieldtrip to use) in case they don't have the signal processing toolbox.

I suspect that in this case @droe11 and @gbShaw used the EEGLAB GUI and I would not expect that to result in `fieldtrip/ft_resampledata` being called. Might it be that EEGLAB is calling the resample function with a 4th argument `h`? It seems that the MathWorks resample takes different inputs than the Octave one from the 4th parameter onward.

But more general: the `fieldtrip/external/signal` directory should not have been on the path to start with.

arnodelorme, 2023-06-26:
I think this is an example of adding all the folders recursively.

gbShaw, 2023-06-26:
> I think this is an example of adding all the folders recursively (manually adding the path)

following commod will add `fieldtrip/external/signal` to matlab search path

`ft_defaults`

However, I am not sure when will `fieldtrip/external/signal/resample.m` will be used as the function of `resample`.

arnodelorme, 2024-02-09:
Fixed. Added extra checks.

closed 2024-02-09 by arnodelorme