sccn/eeglab #636: Can't retain reference when re-referencing to average

A-J-Hanson, 2023-06-21:
The current EEGlab 2023 fails to find the reference channel information when re-referencing even when it has been entered through the channel locations interface.
From what I can tell...
The updated pop_reref.m appears to be looking for chaninfo.removedchans for the reference information. Which isn't created when appending the reference channel infomation.
When previously in EEGlab 2021.1, pop_reref looked for this information in chaninfo.nodatachans. which is created when appending with the reference channel info.

arnodelorme, 2023-06-26:
Good observation. We will fix it.

estebandleonc, 2023-08-24:
Hello, I am having the same problem with this new version 2023 of EEGLAB. Please @arnodelorme could you help us fixing this? Thank you

arnodelorme, 2024-02-09:
This has been fixed in the current version. Thank you.

closed 2024-02-09 by arnodelorme