sccn/eeglab #638: Can not open the signal resource.\n (mff_importsignal() at line 218)

JeongeunY00, 2023-06-26:
Hello.

I have an issue at importing mff file to EEGLAB.

When I choose to import my mff data, it says:

EEGLAB error in fuction mff_importsignal() at line 218:

mff_importsignal (218 lines) 
Error: Can not open the signal resource.\n

I checked that EEG signals are measured well, when I was collecting EEG data.

However the file can have problems.. maybe.

I want to know whether my file is fault or not.

If not, how can I fix this error?

Heartful thanks in advance. 

<br>
#### Versions

|   |   |
|----| ---- | 
| OS version     | window 10 |
| Matlab version | R2021a |
| EEGLAB version | latest |

JeongeunY00, 2023-06-27:
[230626 HW 500-2_20230626_030828.mff.zip](https://github.com/sccn/eeglab/files/11877650/230626.HW.500-2_20230626_030828.mff.zip)
Of course!

arnodelorme, 2024-02-09:
Closing the bug here, but will address it on the other repo this month.

closed 2024-02-09 by arnodelorme